themactep/thingino-firmware #52: T31X gc4653 not working. encoder init FAIL

marcipan73, 2024-04-21:
I use precompiled image "thingino-t31x_gc4653_eth.bin"

```
[root@thingino-t31x ~]#  prudynt
[INFO:main.cpp]: PRUDYNT Video Daemon: Apr 17 2024 07:53:46_1ad8b2756e
[INFO:main.cpp]: Starting Prudynt Video Server.
[INFO:IMP.cpp]: LIBIMP Version IMP-1.1.6
[INFO:IMP.cpp]: SYSUTILS Version: SYSUTILS-1.1.6
[INFO:IMP.cpp]: CPU Information: T31-X
[INFO:IMP.cpp]: Sensor: gc4653
---- FPGA board is ready ----
  Board UID : 30AB6E51
  Board HW ID : 72000460
  Board rev.  : 5DE5A975
  Board date  : 20190326
-----------------------------
VBMCreatePool[335] poolid = 0 IMP_ALLOC failed size = 11059200
[ERROR:Encoder.cpp]: IMP_FrameSource_EnableChn() == -1
[ERROR:main.cpp]: Encoder initialization failed.

```

[root@thingino-t31x ~]# soc -m
t31x

[root@thingino-t31x ~]# sensor all
name:      gc4653
chip_id:   0x4653
i2c_addr:  0x29
width:     2560
height:    1440
min_fps:   5
max_fps:   30
version:   H20211211a

[root@thingino-t31x ~]# dmesg
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.14__isvp_swan_1.0__ (buildroot@buildroot) (buildroot-gcc-13.2.0) #1 PREEMPT Wed Apr 17 07:53:46 UTC 2024
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 RESET ERROR PC:801A7F40
[    0.000000] [<801a7f40>] 0x801a7f40
[    0.000000] CPU0 revision is: 00d00100 (Ingenic Xburst)
[    0.000000] FPU revision is: 00b70000
[    0.000000] cgu_get_rate, parent = 1392000000, rate = 0, m = 129, n = 255, reg val = 0x081000ff
[    0.000000] cgu_get_rate, parent = 1392000000, rate = 0, m = 129, n = 255, reg val = 0x081000ff
[    0.000000] CCLK:1392MHz L2CLK:696Mhz H0CLK:200MHz H2CLK:200Mhz PCLK:100Mhz
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 003e1000 @ 00010000 (usable)
[    0.000000]  memory: 0002f000 @ 003f1000 (usable after init)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 06300000 @ 00000000 (usable)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x062fffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x062fffff]
[    0.000000] On node 0 totalpages: 25344
[    0.000000] free_area_init_node: node 0, pgdat 803eaaa0, node_mem_map 81000000
[    0.000000]   Normal zone: 198 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 25344 pages, LIFO batch:7
[    0.000000] Primary instruction cache 32kB, 8-way, VIPT, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 8-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] pls check processor_id[0x00d00100],sc_jz not support!
[    0.000000] MIPS secondary cache 128kB, 8-way, linesize 32 bytes.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 25146
[    0.000000] Kernel command line: mem=99M@0x0 rmem=29M@0x6300000 console=ttyS1,115200n8 panic=10 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=jz_sfc:256k(boot),64k(env),1344k(kernel),4064k(rootfs),-(rootfs_data)
[... 187 lines cut ...]
  164 root     [khubd]
  195 root     [kworker/0:1]
  196 root     [rpciod]
  199 root     [kworker/0:2]
  204 root     [kswapd0]
  253 root     [fsnotify_mark]
  255 root     [nfsiod]
  263 root     [crypto]
  379 root     [deferwq]
  383 root     [kworker/0:1H]
  395 root     [jffs2_gcd_mtd4]
  442 root     /sbin/syslogd -n -C64 -t
  690 root     /sbin/rngd -r /dev/urandom
  767 root     udhcpc -x hostname:thingino-t31x -S -T1 -t15 -R -b -O search -r
  805 root     /sbin/ntpd -n
  816 root     /usr/sbin/dropbear -k -K 300 -b /etc/issue.net -R
  826 root     /usr/sbin/httpd -f -c /etc/httpd.conf -r Authentication
  836 root     /sbin/crond -f -c /etc/crontabs
  898 root     [irq/37-isp-m0]
  900 root     [irq/38-isp-w02]
 1045 root     -sh
 1047 root     /sbin/watchdog -T 60 /dev/watchdog
 7008 root     /usr/sbin/httpd -f -c /etc/httpd.conf -r Authentication
 7010 root     {mjpeg.cgi} /bin/sh mjpeg.cgi
 7075 root     ps ax

marcipan73, 2024-04-21:
Memory FIX
Kernel command line: mem=65536K@0x0 rmem=65536K@0x4000000

stream now  working 2560x1440 :)  

but no audio......

gtxaspec, 2024-04-22:
@marcipan73 glad to hear you resolved the stream issue.

regarding audio, are you referring to audio output, or audio in via the rtsp stream?

kohalmi666, 2024-04-22:
audio in via the rtsp stream from camrea mic. 
the RTSP stream it contains only the h265 video stream. it does not include the audio stream.

gtxaspec, 2024-04-22:
audio via RTSP is not supported yet, but work is being done to add it to the streamer, stay tuned

gtxaspec, 2024-04-22:
issue #54 created to improve environment handling with regards to rmem

themactep, 2024-04-28:
is this still a thing? my t31x+gc4653 work fine. please close resolved issues.

gtxaspec, 2024-05-05:
Working locally.  Reopen if issues persistent.  😎

closed 2024-05-05 by gtxaspec