themactep/thingino-firmware #54: include rmem with environment configs

gtxaspec, 2024-04-22:
The current U-Boot environment is configured with default memory (rmem) values optimized for 1080p resolutions. This setup might not be ideal for sensors with resolutions that differ from 1080p. When the sensor resolution is either lower or higher than 1080p, this can result in either excessive memory usage or insufficient memory allocation, respectively. This leads to system instability or even prevents the streamer from starting.

We should define and include recommended rmem values in the device profiles for sensors with their respective resolutions. This approach will ensure more efficient memory usage and enhance system stability across different sensor configurations.

gtxaspec, 2024-09-30:
maybe default values based off sensor resolution/sensor?

gtxaspec, 2024-12-21:
added ISP reserved ram to config.soc.in; need to implement logic to update the env on build.

wltechblog, 2025-01-22:
I believe this is done?

gtxaspec, 2025-03-03:
RMEM, ISPMEM,and RMEM implemented as of 03/02.

closed 2025-03-03 by gtxaspec