themactep/thingino-firmware #98: Wifi not working on latest Cinnado D1 build

pauln, 2024-06-02:
Upgrading from a build from about a month ago, via either `sysupgrade -p` or by placing the full image on the SD card as `autoupdate-full.bin`, results in the camera not connecting to wifi.  I see there was a recent change to [use external firmware binaries for the atbm60xx](https://github.com/themactep/thingino-firmware/commit/35b4ee49f84ca2cc10d00056384d2df24d840c1c) - could this be related to that change?  I've checked that my uEnv.txt is up-to-date with [the version in this repository](https://github.com/themactep/thingino-firmware/blob/master/environment/cinnado-d1-t31l.uenv.txt) (having previous had a similar issue when the `wlandev` driver name was changed).  I haven't had a chance to check for any relevant serial output yet; I've restored the previous build in order to get the camera back in service for now.

themactep, 2024-06-02:
Perform a full upgrade with `sysupgrade -f`. I just upgarded mine, everything works fine. Be aware that camera will start an access point, connect to it an open http://thingino.local/ to provision the camera to the new firmware and connect to your local wireless network.

pauln, 2024-06-02:
Thanks - `sysupgrade -f` seems to have done the trick.

closed 2024-06-02 by pauln