themactep/thingino-firmware #112: Add a reset daemon

jayfan0, 2024-06-19:
Reset is possible through the webUI however a reset daemon that monitors internal buttons should also be able to trigger a reset, like stock firmware does.

Can possibly link to the gpio_button env variable

gtxaspec, 2024-06-20:
we should try and leverage the "gpio-keys" part of the linux kernel to do this

gtxaspec, 2024-06-23:
I write and tested a kernel module to initialize the gpio-keys system in the kernel, and I can see the button press events in /dev/input/eventX. However, evtest and thd do not recognize the keycodes as valid. Specifically, when assigning KEY_ENTER (which should correspond to key code 28), evtest reports it as key code 26287, which is invalid and doesn't make sense. I'm still investigating the cause of this issue.