themactep/thingino-firmware #113: gpio_speaker needs active high/low configuration

jayfan0, 2024-06-19:
current env variable `gpio_speaker` seems to either be disabled or active high with the audio module.

current `/etc/modules.d/audio` is
```
audio spk_gpio=$(fw_printenv -n gpio_speaker || echo -1)
```

need to add logic with either `o` or `O` with the `gpio-speaker` definition to make the parameter as follows:

```
audio spk_gpio=$(fw_printenv -n gpio_speaker || echo -1) spk_level=0
```
where:
```
-1 = disabled
0 = active low
1 = active high
```

gtxaspec, 2024-06-20:
fixed in https://github.com/themactep/thingino-firmware/commit/35ad8bf0bbccbada43a75666bebecbb1e0829f08

jayfan0, 2024-06-21:
Thank you for this.

I can confirm that it works.

Both functionally and electrically (with a meter on the amplifier enable pin) for my camera, pin 63 is correctly set on kernel boot:

gpio_speaker 63 = amplifier is set high (3.3v)
gpio_speaker 63O = amplifier is set high (3.3v)
gpio_speaker 63o = amplifier is set low (0v)

jayfan0, 2024-06-21:
update: 

Not defining any pin in gpio_speaker does not automatically set to `-1`. 
Defaults to 63 active high

gtxaspec, 2024-06-22:
fixed in https://github.com/themactep/thingino-firmware/commit/d805d5495dcea30e2d784d35c7b535eae40b631b

default is set to -1 if no gpio is specified in ENV

jayfan0, 2024-06-24:
tested and works fine. thank you

gpio_speaker 63 = amplifier is set high (3.3v)
gpio_speaker 63O = amplifier is set high (3.3v)
gpio_speaker 63o = amplifier is set low (0v)
gpio_speaker <undefined> = gpio 63 is free to manually toggle

closed 2024-06-24 by jayfan0