themactep/thingino-firmware #115: portal issues when accessed through ethernet

jayfan0, 2024-06-21:
The initial configuration portal is designed to be accessed via an access point on a different ip to the normal camera operation.
 
However if the device has an ethernet connection at e.g. 192.168.1.15, the first access to this ip opens the portal. but this is causing issues between a normal firefox browser window and a private window.

The portal should be skipped altogether if a valid ethernet connection is active or the portal is not accessed through its own IPaddress/ network

gtxaspec, 2024-06-23:
@themactep what do you think?  should we have an option to disable portal for wired + wireless devices? or some other switch somewhere
maybe an option when eth0 + wlan0 is detected that pops up on the portal to disable wifi + reboot?

themactep, 2024-06-23:
i think that portal should only be available via the dedicated access point. otherwise, changes can be done from a regular web ui over ethernet.

themactep, 2024-06-23:
check for ethernel in S41portal and exit if found. i think that's the most proper solution.

themactep, 2024-06-23:
check this out
https://github.com/themactep/thingino-firmware/compare/master...no-portal-with-eth

themactep, 2024-06-23:
i do not have a dual port camera to test, so please test it out and report back.

themactep, 2024-06-24:
```
Jun 18 16:39:40 thingino-t21n daemon.info portal[1254]: Interface 'eth0' exists.
```

So the portal detected presence of eth0 and refused to start. But since there is no cable, there is no connection.

We need to check not only for the interface presence but also for it being active. And maybe utilize priority of ports if our little system smart enough to do that.

jayfan0, 2024-06-24:
A few things of notice that may provide some ideas:

1. `eth0` interface was still available even when `disable_eth` was set to true. Maybe this interface needs to be dropped on boot based on that env variable? This makes most sense imo since `wlan0` is able to be dropped and added based on `wlandev`
2. `eth0` does not have an allocated ip address (At least via dhcp, didnt check static) when disabled, so maybe look for an ip?
3. obviously there is no valid ethernet network so check if a ping to local host or gateway or something (taking in to account users who block internet access to cameras via vlan or firewalls)
4. something else

gtxaspec, 2024-06-24:
i believe you can use ` /sys/class/net/eth0/carrier` to determine if the eth0 link is online

jayfan0, 2024-06-24:
i just checked `grep "" /sys/class/net/eth0/*` which includes the output from `carrier`.
It works for detecting if the wire is physically plugged in (`1`) or not (`0`), but if i use `disable_eth=true` then i dont get an ip address but the output is exactly the same

jayfan0, 2024-06-24:
quick update:

This is just a test but I've added a check for the `disable_eth` variable which, if set to `true`, ended up showing the thingino AP, so we're half way there!

```
iface_exists() {                                                                 
        iface="$1"                                                               
                                                            
        if [ -z "$iface" ]; then                            
                echo "Interface parameter missing." 2>&1 | log
                return 1                                    
        fi                                                          
                                                                    
        if ip link show $iface &> /dev/null; then                   
                if [ "eth0" == $iface ] && [ $(get disable_eth) == "true" ]; then
                        echo "eth0 env variable is disabled" | log
                        return 1
                fi                            
                                                  
                echo "Interface '$iface' exists." 2>&1 | log
                return 0                          
        else                                      
                echo "Interface '$iface' does not exist." 2>&1 | log
                return 1
        fi              
} 
```
Now just need to stop the portal running if `wlandev` is defined/ wlan is enabled and ethernet is active

themactep, 2024-06-29:
fixed with #118

closed 2024-06-29 by themactep