themactep/thingino-firmware #142: Wyze v3 + PoE to USB adapter not working

calimansi, 2024-08-03:
Hi,

I bought one of these:  https://a.co/d/9VitpCo

It works with wz mini hacks. I can't get it to work with thingino. 

Wifi works. Ethernet does not work. There is no eth0 device.

gtxaspec, 2024-08-04:
you need to build your own config, and enable the driver, add these to your config:
```
BR2_PACKAGE_USBNET=y
BR2_PACKAGE_USBNET_CDC_ETHER=y```

gtxaspec, 2024-08-16:
any progress?

calimansi, 2024-08-17:
Yes!  It works beautifully.

One more thing though.  I have the uctronics ethernet adapter from amazon.  I noticed that the camera would run for a couple hours then the networking side would crash.  I had to use a realtek program to fix.  Just wanted to reference this link:

https://github.com/gtxaspec/wz_mini_hacks/discussions/495

gtxaspec, 2024-08-17:
great, thanks for referencing that!  Hopefully it will help others.  
Marking as resolved!

closed 2024-08-17 by gtxaspec

tismofied, 2024-11-05:
> Yes!  It works beautifully.
>
>
>
> One more thing though.  I have the uctronics ethernet adapter from amazon.  I noticed that the camera would run for a couple hours then the networking side would [...]

Could you please share your steps how you did this? I have the same adapter