themactep/thingino-firmware #153: Please add options to format the SD card with fat32 or exfat to webui

frogedude, 2024-08-10:
Please add options to format the SD card with fat32 or exfat to webui.

weirdal3333, 2024-08-18:
I think that we should refrain from using exfat. Fat32 is more robust, and it's only downside is that it doesn't support filesizes above 4gb
We can split a file into multiple to conform to that 4gb limit with a script smaller than the exfat drivers.

Additionally, if there's a sudden power loss, fat32 won't randomly be permanently corrupted like exfat can be.

gtxaspec, 2024-08-18:
both vfat and exfat are currently supported, and can be formatted as such on device.  in addition you can also add support for ext2/4, f2fs, etc via buildroot.

invisible999, 2024-08-22:
> I think that we should refrain from using exfat. Fat32 is more robust, and it's only downside is that it doesn't support filesizes above 4gb 

I can easily format SD cards of 64-128GB size with FAT32 under linux. FAS32 limitation is 4TB.

themactep, 2024-08-25:
please check updated webui sd card formatting and confirm it works for you.

closed 2024-08-27 by themactep