themactep/thingino-firmware #191: [feature] Add custom additionnal time for day/night option

Sigri44, 2024-09-01:
Hey, for "Dusk to Dawn", can we obtain two new field ?

- "diff to day", and
- "diff to night"

Because value has not good for my installation, and think can alterate official time, can be more customisable 🙂
Like "+ 1 h 15 min"

Ex : 
My auto time set day at 7 AM, but sun is here at 8:15 AM, so picture still stuck :

[image]

[image]

[image]

themactep, 2024-09-02:
https://github.com/themactep/thingino-firmware/commit/67ae80ab7b6cf82885844c51cc33bace400081a7
please test it out and let me know if any issues.

Sigri44, 2024-09-06:
> [67ae80a](https://github.com/themactep/thingino-firmware/commit/67ae80ab7b6cf82885844c51cc33bace400081a7) please test it out and let me know if any issues.

Done, but don't know hot to test it

[image]

[image]

[image]

themactep, 2024-09-09:
just upgrade the image, all changes are already there.

themactep, 2024-09-20:
please reopen if you have issues with the updated script.

closed 2024-09-20 by themactep