themactep/thingino-firmware #200: Starting wireless

spstrademark, 2024-09-08:
Hello,

i am using the T31L processor with 8 MB flash and cinnado d1.
Got the precompiled bin _thingino-cinnado_d1_t23n.bin_ from here :

[](https://github.com/themactep/thingino-firmware/releases/tag/firmware)
After flashing the bin i get a message " Starting wireless " and the device stops working there.
I have attached a screenshot with the recieved message.
Thank you for your response!

Best regards,
Stanko
[image]

gtxaspec, 2024-09-09:
@spstrademark wrong image, you are using the cinnado firmware for the t23 soc, check the file name.  
you need the T31 image.

themactep, 2024-09-13:
How did you set up the wireless access? From shell over uart or from camera access point and the portal?

spstrademark, 2024-09-17:
Hello,

both ways do not work. The portal doesn't show up. Through the shell, settings can be made, but the camera does not connect to router.
Thank you for your time.

yangminglong, 2024-09-18:
> Hello,
> 
> both ways do not work. The portal doesn't show up. Through the shell, settings can be made, but the camera does not connect to router. Thank you for your time.

Can you provide more information

wltechblog, 2024-09-20:
Does the interface show up when you run iwconfig?

themactep, 2024-09-26:
need more info

closed 2024-09-26 by themactep

spstrademark, 2024-09-26:
Hello,

i have attached the log from the camera
[firstboot.txt](https://github.com/user-attachments/files/17144460/firstboot.txt)

[image]
[image]
[image]

and screenshots what i am recieving 

Thank you for your time!

themactep, 2024-09-26:
Please download a fresh firmware build for your camera, save it on an sd card as `autoupdate-full.bin`, insert it into the camera and reboot with it. Three weeks old build is not something worth investigating. Most likely the error was fixed while ago. If you still have that error after upgrade, then we will start testing it.

spstrademark, 2024-10-01:
Hi, 

the problem persists. But now when trying to connect to wireless there is a blue led blinking. I am flashing the firmware with the usb cloner and it gets stuck on starting wireless.
iwconfig is saying there is no such interface loaded.
[image]

modprobe -l returns atbm6031.ko
[20240930-thingino-log.txt](https://github.com/user-attachments/files/17199605/20240930-thingino-log.txt)
[20240930-thingino-log-failsafe.txt](https://github.com/user-attachments/files/17199606/20240930-thingino-log-failsafe.txt)
[20240930-thingino-log-firstboot.txt](https://github.com/user-attachments/files/17199607/20240930-thingino-log-firstboot.txt)

Thank you for your time!

gtxaspec, 2024-10-01:
please run `thingino-diag -l` and attach the logs

spstrademark, 2024-10-02:
Hello,

i have attached the log file.

Thank you for your time!

[20241002-thingino-diag-log.txt](https://github.com/user-attachments/files/17219689/20241002-thingino-diag-log.txt)