themactep/thingino-firmware #204: Web UI - time sync

wes58, 2024-09-20:
I have thingino firmware installed on 2 wyze v3 cameras.
I have tried to sync time using web interface. But when I press a button to sync time  (on both cameras) I get "Synchronization failed!"
How can I set the time?

themactep, 2024-09-20:
are cameras online and have access to the global internet? what are your ntp servers set to?

wes58, 2024-09-20:
Since I installed the firmware just yesterday I didn't have a lot of time to play with them - I had a hard time to flush the firmware to one of the cameras. But cameras are on line and I can access them from outside the local network (I use no-ip to get the modem external ip address and I can see rtsp streams), so I assume that they can access it as well.
Ntp servers are set, as they were (default) configured in the web UI. I haven't changed anything - yet. 
How does web UI time sync work? Does it use PC time and sends it to the webcam? Or does it trigger camera to request it from ntp server?

wes58, 2024-09-20:
Ok, problem solved. I had to change the gateway ip address and now everything works fine

closed 2024-09-20 by themactep