themactep/thingino-firmware #207: Upgrade firmware

jsissom, 2024-09-21:
I tried to upgrade my firmware and seem to have bricked my device.  It's a Wyze Camera 3 using thingino-wyze_c3_t31x_rtl.bin.

It had been running this version: BUILD_ID="master+aa093d1, 2024-07-19 23:13:50 -0700"

I copied the latest file from 9/19/24 to the SD card in place of the old firmware.  After that, the device doesn't do anything.  I tried reformatting the SD card and going through the install process again and got no response from the device.

Is there a way to start over and get it working again?

Thanks
Jay

jsissom, 2024-09-21:
I was able to unbrick the camera.  I removed the SD card, turned off the camera, turned it on while holding the button and held the button until the blue light came on.  After waiting a few minutes, it started advertising it's Wifi AP and was able to initialize it.  It was still using the old firmware.

Is there a way to upgrade the camera to the newer firmware?  I have some on new firmware and some on old and the URLs have changed so would like to get all of them on the new firmware if it is possible.

Thanks
Jay

wltechblog, 2024-09-21:
You can always update the firmware to the latest version for your device via ssh.
https://github.com/themactep/thingino-firmware/wiki/Updating-Firmware#from-the-camera-itself

jsissom, 2024-09-22:
Thank you.  I had never ssh'ed into the camera.  I was able to update them this way.  I found that the root password got reset to default but I was able to set it back after the update.

Thanks for your hard work on this.  This firmware has made my cameras much more useful to me.

closed 2024-09-22 by jsissom