themactep/thingino-firmware #217: T31X GC4653 ethernet

kaar3l, 2024-09-25:
Hello I did buy 4 cameras with T31 + GC4653. I removed flash chip and installed thingino-vanhua_z55i.bin firmware. First 2 cameras were perfect, after flashing booted up and ethernet was working like charm. Next two cameras I flashed with exactly the same firmware and nothing. The ethernet doesn't work. The boards seem identical.

These are the 2 cameras that are not working:
[image]
[image]

I did solder serial to the cameras and seems that the cameras boot up but no ethernet connection is made.

gtxaspec, 2024-09-25:
can you take a photo of the bottom of the board?

kaar3l, 2024-09-25:
Not the best quality but here it is.
[image]
I did not compare the other side of the camera board with the working one.

themactep, 2024-09-25:
these modules are better flashed over serial port. login to uboot, use ymodem to upload the image. it takes time but it's safer than removing the flash chip. try to flash stock back and confirm that ethernet works. if it does not - the module is damaged during soldering. phy chip is located on the opposite side of the flash chip, between two uarts.

themactep, 2024-09-25:
[image]

This one is on my bench right now. Ethernet works with thingino ootb.

themactep, 2024-09-26:
please feel free to reopen if you find a confirmation that the issue is in the firmware.

closed 2024-09-26 by themactep