themactep/thingino-firmware #219: Xiaomi iSC5 sound issues

tERB00, 2024-09-26:
First of all thank you to all Thingino devs for your hard work!

I just installed the firmware on my old Xiaomi iSC5 and it works fine out of the box except few small things.

One of them is audio. Audio playback works but it plays audio very fast (probably compile error).
But the audio input (mic) does not work at all. I can see audio track on the rtsp stream, changing the codec changes the stream's audio codec too but there is no audio. Complete silence. No hiss, no humm - nothing. Enabling AGC does not affect anything as well.

I would like to help in debugging that.

tERB00, 2024-09-27:
My bad - no mic was a VLC issue, sorry about that.
Fast playback is still an issue tho.

themactep, 2024-09-30:
make sure you have speaker gpio pin defined and it is set to power up the amplifier ic.

tERB00, 2024-09-30:
I didn't compiled the firmware but there is
_gpio_speaker=63_
in uenv

gtxaspec, 2024-09-30:
does fast playback happen in other players, or just VLC? try MPV/go2rtc/etc to test.

tERB00, 2024-09-30:
Oh. Fast playback is in the camera's speaker, not in the player

closed 2024-10-12 by tERB00