themactep/thingino-firmware #220: Xiaomi iSC5 Local Record turns itself off

tERB00, 2024-09-27:
Another issue with Xiaomi iSC5 is non functioning Local Record.
It turns itself off after a minute (the actual setting option in /etc/webui/record.conf)

The only thing related to the record manager in the logs I found is this:

```
Sep 27 12:57:00 Xiaofang cron.info crond[1531]: USER root pid 22829 cmd daynight
Sep 27 12:57:44 Xiaofang daemon.info openRTSP[23510]: Stopping openRTSP
Sep 27 12:57:44 Xiaofang daemon.info openRTSP[23510]: no process in pidfile '/run/blink.pid' found; none killed
Sep 27 12:57:44 Xiaofang daemon.info openRTSP[23510]: OK
Sep 27 12:57:44 Xiaofang daemon.info openRTSP[23510]: no process in pidfile '/run/record-manager.pid' found; none killed
Sep 27 12:57:44 Xiaofang daemon.info openRTSP[23510]: OK
Sep 27 12:57:44 Xiaofang daemon.info openRTSP[23510]: start-stop-daemon -K -n openRTSP
Sep 27 12:57:44 Xiaofang daemon.info openRTSP[23510]: no openRTSP found; none killed
Sep 27 12:57:44 Xiaofang daemon.info openRTSP[23510]: OK
Sep 27 12:57:45 Xiaofang daemon.info openRTSP[23510]: Starting openRTSP
Sep 27 12:57:47 Xiaofang daemon.info openRTSP[23510]: OK
Sep 27 12:57:47 Xiaofang user.notice root: Used disk space is 0%
Sep 27 12:57:49 Xiaofang daemon.info openRTSP[23510]: start-stop-daemon -b -S -x openRTSP -- -u   -w  -h  -f  			-P 160 -i -b 1048576 -t -F thingino- rtsp://127.0.0.1/ch0
Sep 27 12:57:49 Xiaofang daemon.info openRTSP[23510]: OK
Sep 27 12:57:58 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: Active motion detected in region 0
Sep 27 12:57:58 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: Motion Start
Sep 27 12:58:00 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: Active motion detected in region 0
Sep 27 12:58:00 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: Active motion detected in region 0
Sep 27 12:58:00 Xiaofang cron.info crond[1531]: USER root pid 24039 cmd daynight
Sep 27 12:58:00 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: Active motion detected in region 0
Sep 27 12:58:02 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: Active motion detected in region 0
Sep 27 12:58:02 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: End of Motion
Sep 27 12:58:07 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: Active motion detected in region 0
Sep 27 12:58:07 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: Motion Start
Sep 27 12:58:09 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: Active motion detected in region 0
Sep 27 12:58:15 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: Active motion detected in region 0
Sep 27 12:58:16 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: End of Motion
Sep 27 12:58:32 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: Active motion detected in region 0
Sep 27 12:58:32 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: Motion Start
Sep 27 12:58:34 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: Active motion detected in region 0
Sep 27 12:58:35 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: Active motion detected in region 0
Sep 27 12:58:36 Xiaofang user.info prudynt[1741]: [INFO:Motion.cpp]: End of Motion
Sep 27 12:58:47 Xiaofang user.notice root: openRTSP is not running. Shutting down recorder
Sep 27 12:58:47 Xiaofang daemon.info openRTSP[24884]: Stopping openRTSP
Sep 27 12:58:47 Xiaofang daemon.info openRTSP[24884]: no process in pidfile '/run/blink.pid' found; none killed
Sep 27 12:58:47 Xiaofang daemon.info openRTSP[24884]: OK
Sep 27 12:58:47 Xiaofang daemon.info openRTSP[24884]: stopped process in pidfile '/run/record-manager.pid' (pid 23614)
Sep 27 12:58:47 Xiaofang daemon.info openRTSP[24884]: OK
Sep 27 12:58:47 Xiaofang daemon.info openRTSP[24884]: start-stop-daemon -K -n openRTSP
Sep 27 12:58:47 Xiaofang daemon.info openRTSP[24884]: no openRTSP found; none killed
Sep 27 12:58:47 Xiaofang daemon.info openRTSP[24884]: OK
Sep 27 12:59:00 Xiaofang cron.info crond[1531]: USER root pid 25228 cmd daynight
Sep 27 13:00:00 Xiaofang cron.info crond[1531]: USER root pid 26675 cmd daynight
```

The SD card itself is functional. If I add subfolder to the record path (record_path="/mnt/mmcblk0p1/vids/") the folder is being created as it should.

nicolasroche898, 2024-10-13:
It's missing all the values:

start-stop-daemon -b -S -x openRTSP -- -u   -w  -h  -f  -P 160 -i -b 1048576 -t -F thingino- rtsp://127.0.0.1/ch0

It's supposed to be -u thingino thingino -w 1980 -h 1080 -f 30 ...

gtxaspec, 2024-10-13:
Local record is broken right now, as we refactor the config logic. Stay tuned.

PR #252 (2024-10-13): prudynt.cfg contains a mix of k:v and k=v, support both fmts
prudynt.cfg contains a mix of both
key: value
and
key = value

support both formats, with optional whitespace before and after separator, as they seem to be added by prudynt upon parameter update

fixes #220
closed 2024-10-13 by themactep