themactep/thingino-firmware #224: Wyze Cam 3 image and mjpeg only work if preview is running

ClearAirTurbulence3D, 2024-09-29:
I may not be understanding something...
If I issue a request for an image (with the proper login info) via http://ip/x/image.cgi
I get an image. _If I repeat the command_, I get the same image, even though the timestamp in the file name changes.

The same happens with http://ip/x/mjpeg.cgi - just a frozen frame with the last image taken.

If I open a tab on the browser with http://ip/x/preview.cgi I can get updated images and mjpeg, as long as that tab/connection is open. 

Shouldn't the camera update the image or the mjpeg with a new frame every time the request is made?

gtxaspec, 2024-09-30:
how are you retrieving the image? i just tested the same endpoint, image.cgi in chrome, and it gave me two different files, with different names, the images different (verified by osd timestamp)

any errors in the logs?

ClearAirTurbulence3D, 2024-10-01:
I figured out the problem - I updated the firmware to the latest and it works as expected. The previous firmware was from early Sept.

closed 2024-10-01 by ClearAirTurbulence3D