themactep/thingino-firmware #231: how to determine which Wyzecam V3 variant you have

goatzillax, 2024-10-01:
I'm still running wz_mini_hacks but I think I'm kind of ready to jump ship as of the recent forced update.

How do I figure out which of the 5 apparent variants my V3 is?

lsmod shows:

```
sensor_gc2053_t31 11314 0 - Live 0xc0844000 (O)
8189fs 1601252 0 - Live 0xc05f0000 (O)
```

That seems to narrow it down to two, however what's the difference between T31X and T31ZX?

Is it possible to figure it out without cracking the camera open?

tERB00, 2024-10-01:
You have to backup your original firmware anyways before doing anything else. So try out and see if Ingenic USB Cloner works. If not you have to open it and hook up UART.

gtxaspec, 2024-10-01:
z and zx are interchangeable, the big difference is T31A or Z/ZX.  run this command in the shell and share the output:
`devmem 0x13540238`

goatzillax, 2024-10-01:
OK great.  The raw devmem gave:

```
# devmem 0x13540238
0x66661111
```

And that script returned: 

```
t31zx
```

So it seems the firmware I want is "T31ZX, GC2053, RTL8189FTV, 16MB".  Any other low-hanging fruit I should look at?

Thanks!

themactep, 2024-10-01:
I advise joining our [Telegram group](https://t.me/thingino) and [Discord server](https://discord.gg/BTrPtSEN) for support questions. Please leave GH Issues for issues. Thank you.

closed 2024-10-01 by themactep