themactep/thingino-firmware #239: wyze_vdb1: can no longer pair chime

derzahla, 2024-10-06:
Theres been some discussion on discord about this and I know @gtxaspec said he would look into this, so just wanted to open an issue for tracking. 

My chime would lose pairing after a week or so but for the last 3weeks or so I have been unable to get the chime to repair at all.  I have used an extension cord to place the chime right next to my door bell, run 'doorbell_ctl -p XX:XX:XX:XX`;sleep Y' in a loop while repeatedly initiating pairing-mode on the chime and power cycling frequently, total testing times have totaled well over 1 hour and included hard power resets of the doorbell itself and firmware update attempts.  

Note: 'sysupgrade -p' as recent as last night does not upgrade to a newer build than 'master+ed5d38b, 2024-09-18' - not sure if intentional or not.

gtxaspec, 2024-10-06:
i'll test this again today.

themactep, 2024-10-11:
@derzahla is it resolved with the latest bullds?

derzahla, 2024-10-11:
After updating to 2024-10-4 I was able to pair my chime. I did set 'gpio 61 high' before I tried to pair though. It has been going strong for 1 week.  If yours is also paired and stable, and 'gpio 61 high' is now the default then it's likely resolved.

closed 2024-10-15 by themactep