themactep/thingino-firmware #251: Settings of "Prudynt" can not save

yangminglong, 2024-10-12:
I have updated the latest Thingino firmware (2024.10.10)

1. I modified the “Main stream” - “FPS” from 25 to 20, press F5 to refresh, and keep the FPS configuration at 20. After restarting the camera, the FPS was restored to 25

2. I modified the "Main stream" -  "Audio" Switch from the default value of off to on, and refreshed it by pressing F5. The switch was restored to off

ej52, 2024-10-12:
@yangminglong Please upgrade to the latest firmware, all issues related to Prudynt config have been fixed as of `2024.10.12`.

`Prudynt` page has been changed to `Streamer`

yangminglong, 2024-10-13:
> @yangminglong Please upgrade to the latest firmware, all issues related to Prudynt config have been fixed as of `2024.10.12`.
> 
> Also `Prudynt` page has been changed to `Streamer`

Great, I have use `2024.10.12 firmware` tested and both of the issues I mentioned have been fixed. Thank you very much. 

Now I can save the configuration changes of the Main stream and Sub stream normally after restarting the camera.

I found that there is still a problem with the configuration of the Audio group: I saw that the default configuration for Sampling is 16000Hz, but I saw in the RTSP stream of VLC that the codec information displayed Sampling as 48000Hz. I made some modifications to Sampling and then restarted the camera, but Sampling remained at 48000Hz. I am not sure if other Audio configurations can be saved properly.

But it doesn't seem important, I can hear the camera's sound through RTSP streaming, and the sound size is also appropriate

Thank you again, Thingino firmware looks really great now

My camera is:  iflytek_xfp301_m_rtl8188ftv

ej52, 2024-10-15:
@yangminglong I did some testing regarding audio and it does seem that no matter what Sampling is chosen it always defaults to 48000.

The underlying issue is coming from prudynt-t itself, seems like it is disregarding `input_sample_rate` in the config and using the default value set in code.

themactep, 2024-10-15:
@nschimme @gtxaspec @Lu-Fi can you guys please check that out, prudynt seems ignores sampling settings.

[image]

nschimme, 2024-10-15:
Opus always shows up at 48 kHz no matter what the input sampling

Lu-Fi, 2024-10-16:
@nschimme is it possible to provide some information on whicht codec can support which setting ?

nschimme, 2024-10-16:
@Lu-Fi Yeah, it's hard coded here in the `sample_rate` variable from when I was doing manual testing and using ffplay to validate the sound was correct:
https://github.com/gtxaspec/prudynt-t/blob/master/src/IMPAudio.cpp#L97

`G711A`, `G711U`, and `G726` codecs have a mandatory sample rate of 8 kHz. We could just limit the available sample rates for those codecs. `Opus` allows for any sample rate but will still show 48 kHz as part of ffplay metadata. That's just an Opus quirk and nothing we can fix.

gtxaspec, 2024-10-16:
We should have it default to 48khz, but be adjustable in the config to what the user wants, it would be nice =D

nschimme, 2024-10-16:
There are limitations in the codecs that prevent that. The user problem here is that they're surprised and there isn't much we can do other than hide certain sample rates where its mandatory to be 8 kHz for that codec, or in the case of Opus we say that it's working but will have a note show up that 48 kHz will come out.

nschimme, 2024-10-16:
Oh, on the note of codec mandatory settings. `G726` mandates a bitrate of 16:
https://github.com/gtxaspec/prudynt-t/blob/master/src/IMPAudio.cpp#L89

themactep, 2024-10-17:
Is it an IMP imposed limit? Whitepaper states that "G.726 : 40, 32, 24, 16 kbit/s Adaptive Differential Pulse Code Modulation (ADPCM)"
https://www.itu.int/rec/dologin_pub.asp?lang=e&id=T-REC-G.726-199012-I!!PDF-E&type=items

nschimme, 2024-10-17:
Yeah, the Ingenic SDK only provides G.726-16 so we'll have to implement our own G726 if we want the other variants.

themactep, 2024-10-17:
i have added automatic changes to sampling rates when selecting different codecs. please check and verify. let me know if more variants are needed.

gtxaspec, 2024-10-17:
we should still implement variable sample rate in opus.  even if it will still report as 48khz, lowering the samplerate will save bandwidth and maintain quality since opus is pretty good.
[image]

ej52, 2024-10-17:
The Opus library used only seems to allow a sample rate of 48 kHz, 24 kHz, 16 kHz, 12 kHz, 8 kHz.

https://github.com/xiph/opus/blob/7db26934e4156597cb0586bb4d2e44dccdde1a59/src/opus_encoder.c#L546

It seems like the bandwidth profile can be set to save bandwidth as @gtxaspec said.

https://github.com/xiph/opus/issues/170
https://datatracker.ietf.org/doc/html/rfc6716#section-2

virmaior, 2025-02-28:
for recorded files as 2025-02-27, the opus codec with nothing changed will cause ffmpeg to think there's no audio, because of an incorrect header and the inability of ffmpeg to figure it out from the data.

nschimme, 2025-02-28:
@virmaior that might be due to https://github.com/gtxaspec/prudynt-t/pull/71

I believe we can close this ticket as Prudent does observe sampling/bitrate settings now.