themactep/thingino-firmware #254: WebUI Network settings - Use DHCP issue

wes58, 2024-10-14:
I have downloaded the latest firmware on WyzeCam 3 and I have noticed the following issue.
When I go to WebUI and network settings, I select "Use DHCP" to OFF. I set IP and other fields.
When I click "Save Changes" the "Use DHCP" switch goes back to ON.
After rebooting the camera and opening the WebUI network settings page the switch "Use DHCP" is ON.
When I turn it OFF it shows that parameters that I set before, though!

I checked on the second camera with older firmware and the switch "Use DHCP" retains the state that it was set to.

themactep, 2024-10-15:
using static address on a wifi connection is a bad practice in general, pin the mac address to a desired ip address on your dhcp router and that will be equal to a static ip with a twist that you won't be locked out of the network if the target address is already accidentally occupied.

nevertheless, i'll look into the dhcp form later this week.

wes58, 2024-10-15:
I know all that. That how I used to have it set up when I used wz_mini_hacks. But with my configuration I had some problems so when I saw that I can set up static IP with thingino firmware, I decided to do it this way.
Thanks for your response.

themactep, 2024-10-15:
technically, wpa_supplicant can handle multiple networks. but i never tried that on camera, only on a full desktop. here's a little info https://raspberrypi.stackexchange.com/questions/11631/how-to-setup-multiple-wifi-networks

wes58, 2024-10-16:
After reading the information on the link that you gave me, it looks promising. But I can't do anything about it. I don't know much about Linux and I can't compile the firmware.
Thanks

mominulaziz, 2025-02-12:
I have Wyze V3 with running build id: master+f358a8e, 2025-02-04 09:20:24 +0000
I am unable to save Static ip. As soon as I disable DHCP and provide required information I get the error:  wlan0 MAC address format is invalid. Also, my TP link router has a warning for the same device: Random MAC address detected. This device is dropping the wifi like crazy but about 5 feet away from the router. So, I tried to have it as static from the router but still same issue now trying to do it from the web ui but I am unable to do it. I have used the thingino firmware to this device and it was never registered through the wyze app. So, not sure what is happening. I also see a generate random mac option but not sure what and how to use it.

themactep, 2025-02-12:
what is your mac address? i just switched my v3 to a static ip from web ui, no issues. your build might be too old. tru to perform a full upgrade.

mominulaziz, 2025-02-13:
I did a full upgrade and now it is working. The only issue i think is that performing full upgrade is not convenient as I have to setup everything from scratch like wifi, stream password, stream configuration.

themactep, 2025-02-17:
Tonight build will have persisted partition for keeping configurations between upgrades.

closed 2025-02-17 by themactep