themactep/thingino-firmware #259: Wifi AP configuration page can be re-POSTed to from a cache browser window

8none1, 2024-10-19:
I have configured two cameras today.  When I flashed the first one I connected to the provided access point from my iPhone.  I provided the config information and submitted the form.  The device rebooted and I everything worked.

Later on that same day I was flashing a second device.  I had a few snags because it was running Dafang, but in the end I was able to get it working with help from the Discord channel (thank you!).

When I connected my phone to the access point on the second camera and opened up the browser I still had the original config page open at the same URL (http://172.16.0.1).  When the access point connected the browser attempted to refresh the page and I think it re-POSTed the form.

The camera then accepted that form as it's configuration data and rebooted.

The camera came up with the same hostname and root password (and perhaps other things) as the first camera.

This was a mild hold up and easily fixed.

gtxaspec, 2024-10-27:
@themactep we can't expire the data using httpd or via the web, can we?

themactep, 2024-10-27:
the page is in client's browser and any attempt to refresh it will send the same requests as was sent previously. http is a stateless protocol, we cannot control what's on the client's side. i usually make a GET redirect to another url so that there was no pending POST page in the browser. But it did not do that on the portal because most devices just close the portal window after the network is gone. I had hard time even to reproduce the issue. I think I'll extract Done page to a separate html file and will redirect to it.

themactep, 2024-10-29:
updated. please test and confirm it works.

8none1, 2024-10-31:
TL;DR:  Fixed.  Thank you.

I cloned from git, built the image for a Cinnado D1 t23, copied the resulting bin file to the root of an sd card, renamed it `autoupdate-full.bin` and booted the camera with the card in.

The Wifi API appeared, I connected to it and completed the form for config.  All good.

I removed the card, deleted the `.done` file, reinserted the card and booted the camera again.

I was NOT able to reproduce the automatic POSTing of details to the camera.

closed 2024-10-31 by 8none1