themactep/thingino-firmware #261: [Feature] Add custom date string formats to OSD

8none1, 2024-10-20:
I would like to be able to specify a custom `strftime` formatting string so that I can change the way the time is shown when overlaid on the stream.

In my case I want to add the timezone.

I was able to do this by editing the `/etc/prudynt.cfg` file like so:

```text
stream0 : 
{
  osd : 
  {
    time_format: "%F %T %Z"
  };
  fps = 20;
  audio_enabled = false;
};
```
(The default is `%F %T` and so is not set in the config file)

Which results in:
[image]

I'd love to be able to specify this via the WUI as well:

[image]

nicolasroche898, 2024-10-20:
I use time_format = "%F %T %z %a";
shows the day of the week ("Sunday") and the current numerical time offset, which can change between DST and not DST depending on season

themactep, 2024-11-10:
Done.