themactep/thingino-firmware #267: ipv6 compatability checklist

gtxaspec, 2024-10-21:
turn up full ipv6 compatability:

- prudynt
  - listens on ipv4/ipv6 via live555 [ok]
- webui 
  - complains about no network connection when no ipv6 only
- ~~send2termbin~~
  - ~~busybox nc has poor ipv6 support (fixed, switch to curl)~~

probonopd, 2024-12-29:
> complains about no network connection when no ipv6 only

Not entirely sure what you mean by "when no ipv6 only". The whole reason for me to switch to this firmware would be IPv6 support. So is IPv6 (only) possible?

gtxaspec, 2024-12-29:
> > complains about no network connection when no ipv6 only
> 
> Not entirely sure what you mean by "when no ipv6 only". The whole reason for me to switch to this firmware would be IPv6 support. So is IPv6 (only) possible?

Yes, the message issue is only cosmetic.

wltechblog, 2026-02-25:
@gtxaspec is this status still current?

closed 2026-03-16 by themactep