themactep/thingino-firmware #283: bug: wrong 25 fps on newly fully flashed jooan camera

nicolasroche898, 2024-10-26:
repro:
download full firmware for jooan a6m camera, flash full
go to web ui streamer:

seen: 25fps for stream0 and stream1
expected: 15fps for stream0 and stream1

camera doesn't support 25fps.
this creates streaming and recording problems because player doesn't get the correct frame rate.

nicolasroche898, 2024-10-26:
[image]

themactep, 2024-10-26:
Open web console, check initial request and response to populate the form. Does it receive 25 fps from the streamer?

nicolasroche898, 2024-10-26:
yes it receives 25fps from the websocket upon opening the config page. tested on freshly flashed camera:

`{"stream0":{"audio_enabled":true,"bitrate":3000,"buffers":2,"enabled":true,"format":"H264","fps":25,"gop":20,"height":720,"max_gop":60,"mode":"SMART","profile":2,"rotation":0,"rtsp_endpoint":"ch0","width":1280},"stream1":{"audio_enabled":true,"bitrate":1000,"buffers":2,"enabled":true,"format":"H264","fps":25,"gop":20,"height":360,"max_gop":60,"mode":"SMART","profile":2,"rotation":0,"rtsp_endpoint":"ch1","width":640},"audio":{"input_agc_compression_gain_db":0,"input_agc_enabled":false,"input_agc_target_level_dbfs":10,"input_alc_gain":"not supported on this plattform","input_bitrate":40,"input_enabled":true,"input_format":"OPUS","input_gain":25,"input_high_pass_filter":false,"input_noise_suppression":0,"input_sample_rate":16000,"input_vol":80},"image":{"ae_compensation":128,"anti_flicker":2,"backlight_compensation":null,"brightness":128,"contrast":128,"core_wb_mode":0,"defog_strength":null,"dpc_strength":null,"drc_strength":null,"highlight_depress":0,"hue":null,"max_again":160,"max_dgain":80,"running_mode":0,"saturation":128,"sharpness":128,"sinter_strength":128,"temper_strength":128,"wb_bgain":0,"wb_rgain":0}}`

cat /etc/prudynt.cfg
also shows:
fps: 25;
for stream0, not commented

nicolasroche898, 2024-10-26:
(after modifying fps from web ui)
sensor fps is wrong:

```
# grep -e " :" -e fps /etc/prudynt.cfg 
general : 
rtsp : 
sensor : 
  fps = 25;
image : 
stream0 : 
  osd : 
  fps = 15;
stream1 : 
  osd : 
  fps = 15;
stream2 : 
  jpeg_idle_fps = 1;
  fps = 25;
```

themactep, 2024-10-27:
i think 25 is default fps in prudynt. @gtxaspec can we add initialized fps in exported sensor driver values? so that prudynt could get the real values from /proc/jz/sensor/fps ?

nicolasroche898, 2024-11-01:
still occurring on BUILD_ID="master+7328f82, 2024-10-31 06:19:03 +0000"

this causes recorder and streamer to create wrong stream advertising 25fps, which are skipping upon playback

this is a problem also for all 30fps cams such as Wyze V3 default.

themactep, 2025-03-26:
@gtxaspec @Lu-Fi guys we need to make prudynt to config with the proper fps for a given sensor. either presetting it during compilation based on the sensor used or even better reading it from the sensor data in runtime.

themactep, 2025-03-26:
and i see that it does read the value from proc_fs if one is not defined in the config. but it is defined (hence forced) as 25 in the default config. gotta disable it in a local build and recompile, see what breaks.

gtxaspec, 2025-03-26:
This won't be possible until every sensor max fps is checked, the values in the sensor sources are not always accurate. In progress

davidebeatrici, 2025-07-04:
Please note that this also happens with the Jienuo JN-107-AR-E-WIFI, `/etc/prudynt.cfg` has to be edited manually in order for the streamer to work.

```
[   15.579242] probe ok ------->sc5235
[   15.630318] -----sensor_detect: 1005 ret = 0, v = 0x52
[   15.630939] -----sensor_detect: 1013 ret = 0, v = 0x35
[   15.630946] sc5235 chip found @ 0x30 (i2c0)
[   15.630952] sensor driver version H20220914a
[   15.907159] sc5235 stream on
[   15.959053] warn: fps(1638401) not in range
[   15.959070] CPU: 0 PID: 1810 Comm: prudynt Tainted: G           O 3.10.14__isvp_swan_1.0__ #1
[   15.959077] Stack : 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   15.959077]     00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   15.959077]     00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   15.959077]     00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   15.959077]     00000000 00000000 00000000 00000000 00000000 00000000 00000000 82e097f0
[   15.959077]     ...
[   15.959151] Call Trace:[<8001f724>] 0x8001f724
[   15.959165] [<8001f724>] 0x8001f724
[   15.959172] [<c0218078>] 0xc0218078
[   15.959179] [<c02ef060>] 0xc02ef060
[   15.959186] [<c021c334>] 0xc021c334
[   15.959193] [<c02425d8>] 0xc02425d8
[   15.959201] [<c027e578>] 0xc027e578
[   15.959208] [<c02252d8>] 0xc02252d8
[   15.959217] [<c021e6f8>] 0xc021e6f8
[   15.959224] [<800e589c>] 0x800e589c
[   15.959233] [<801bbfa4>] 0x801bbfa4
[   15.959239] [<801bba38>] 0x801bba38
[   15.959247] [<80096c2c>] 0x80096c2c
[   15.959253] [<801bb59c>] 0x801bb59c
[   15.959259] [<80190dcc>] 0x80190dcc
[   15.959265] [<8009bc38>] 0x8009bc38
[   15.959272] [<80096b4c>] 0x80096b4c
[   15.959278] [<8009725c>] 0x8009725c
[   15.959286] [<8009725c>] 0x8009725c
[   15.959293] [<80065c24>] 0x80065c24
[   15.959300] [<80061b14>] 0x80061b14
[   15.959306] [<800663a4>] 0x800663a4
[   15.959312] [<80061bac>] 0x80061bac
[   15.959319] [<80062520>] 0x80062520
[   15.959325] [<801ae254>] 0x801ae254
[   15.959331] [<800666f0>] 0x800666f0
[   15.959337] [<800659a4>] 0x800659a4
[   15.959343] [<802670dc>] 0x802670dc
[   15.959349] [<803ac8d8>] 0x803ac8d8
[   15.959355] [<8008ca00>] 0x8008ca00
[   15.959363] [<80054a68>] 0x80054a68
[   15.959369] [<802e8a88>] 0x802e8a88
[   15.959375] [<8008ee1c>] 0x8008ee1c
[   15.959382] [<80091488>] 0x80091488
[   15.959388] [<800e0728>] 0x800e0728
[   15.959393] [<800dfea4>] 0x800dfea4
[   15.959401] [<8025a87c>] 0x8025a87c
[   15.959407] [<800adb30>] 0x800adb30
[   15.959415] [<800e589c>] 0x800e589c
[   15.959421] [<8025b034>] 0x8025b034
[   15.959427] [<801a8148>] 0x801a8148
[   15.959434] [<8025c8a0>] 0x8025c8a0
[   15.959440] [<800245bc>] 0x800245bc
[   15.959446] [<800d91e8>] 0x800d91e8
[   15.959453] [<80018f6c>] 0x80018f6c
[   15.959459] [<c02209ac>] 0xc02209ac
[   15.959465] [<80069894>] 0x80069894
[   15.959472] [<800dad80>] 0x800dad80
[   15.959478] [<800dbb88>] 0x800dbb88
[   15.959485] [<800574e0>] 0x800574e0
[   15.959491] [<8002249c>] 0x8002249c
[   15.959500] 
[   15.959507] Failed to set sensor fps=1638401, ret=-1
[   15.962267] sc5235 stream off
[   40.400553] RTW: Turbo EDCA =0xa42b
[   42.400553] RTW: Turbo EDCA =0x5ea42b
[   47.921770] RTW: rtw_issue_addbareq_cmd, p=0
[   47.921867] RTW: BA_starting_seqctrl = 160 for TID=0
[   47.921892] RTW: issue_addba_req(wlan0) ra=9c:9d:7e:74:d2:90 tid=0
[   47.923108] RTW: OnAction_back
[   47.923122] RTW: OnAction_back, action=1
[   47.923128] RTW: agg_enable for TID=0
[   50.400533] RTW: Turbo EDCA =0xa42b
[   52.400527] RTW: Turbo EDCA =0x5ea42b
[   60.400476] RTW: Turbo EDCA =0xa42b
[   62.400733] RTW: Turbo EDCA =0x5ea42b
[   82.400380] RTW: Turbo EDCA =0xa42b
```

nicolasroche898, 2025-07-06:
the bug still exists for jooan a6m. you have to manually decrease the frame rate to 15 fps in prudynt to fix it.

davidebeatrici, 2025-07-07:
That may actually be related to #640.

themactep, 2025-09-06:
New prudunt build gets its sensor parameters directly from sensor, not from the config. Please check of that resolved the issue.

themactep, 2025-12-29:
is this still an issue? please do not forget to close resolved tickets.