themactep/thingino-firmware #284: bug: changing fps in the gui doesn't do anything

nicolasroche898, 2024-10-26:
repro:
flash a new jooan camera
observe "25fps" in the streamer config page
move the "fps" scroll bar to 15fps with the mouse. 
web ui shows 15 fps... but settings are saved anywhere:

seen: cat /etc/prudynt.cfg still shows 25fps
expected: cat /etc/prudynt.cfg should show 15fps

themactep, 2024-10-26:
Open console in browser dev tools (f12) and check request and response from prudunt in console log. Do you get any error?

nicolasroche898, 2024-10-26:
changing from 15fps to 14fps in the gui:

```
1729924404270 '===>' '{"stream0":{"fps":14},"action":{"save_config":null,"restart_thread":3}}'
config-streamer.cgi:697
1729924404334 '<===' '{"stream0":{"fps":14},"action":{"save_config":"initiated","restart_thread":"initiated"}}'
```

nicolasroche898, 2024-10-26:
waitasec, now in config file:
fps = 14;

so problems exists only in initial configuration?
i'll reset the camera and try again

nicolasroche898, 2024-10-26:
well that it very weird. now modification works from web gui. i don't know what to say.

nicolasroche898, 2024-10-26:
keep it open, if i can't reproduce in the next few days, i'll close

nicolasroche898, 2024-10-26:
tested again on new camera. couldn't reproduce. closing.

closed 2024-10-26 by nicolasroche898