themactep/thingino-firmware #285: bug: sysupgrade from relative path doesn't work

nicolasroche898, 2024-10-26:
repro:
upload thingino-jooan_a6m_atbm6012bx.bin to /tmp
ssh to device
# cd /tmp
tmp# sysupgrade thingino-jooan_a6m_atbm6012bx.bin 
Upgrade from a local file
Cannot find file thingino-jooan_a6m_atbm6012bx.bin

this works, using full path:
tmp# sysupgrade /tmp/thingino-jooan_a6m_atbm6012bx.bin 
Upgrade from a local file
Upgrading from a provided file
Stop services
...etc

themactep, 2024-10-26:
this is how linux works. working directory is not a part of PATH. prefix file name with `./` to make it relative.

gtxaspec, 2024-10-27:
```
root@ing-jdpan01--4a32 mmcblk0p1# sysupgrade ./thingino-wyze_cp2-update.bin 
Upgrade from a local file
Cannot find file ./thingino-wyze_cp2-update.bin
```

themactep, 2024-10-27:
```
root@ing-wyze-c3-bdca ~# touch aaa
root@ing-wyze-c3-bdca ~# sysupgrade ./aaa
Upgrade from a local file
Cannot find file ./aaa in /tmp/sysupgrade
```

the script is copied to run from /tmp/sysupgrade/ but the fw file stays in its original directory hence `./` is not equal to `pwd` anymore. it works for full path but not for relative.

themactep, 2024-10-27:
i've changed order of validation and isolation, made it move the original fw file to working directory prior to checking. also added upgrading of sysupgrade script from github. please test and report back.

themactep, 2024-10-27:
```
root@ing-wyze-c3-bdca ~# touch aaa
root@ing-wyze-c3-bdca ~# sysupgrade ./aaa 
Found working directory /tmp/sysupgrade
Upgrade from a local file
Switching to working directory /tmp/sysupgrade
/tmp/sysupgrade
Updating self.
curl -fL# -o /tmp/sysupgrade/sysupgrade "https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/overlay/lower/usr/sbin/sysupgrade"
################################################################################################################# 100.0%
curl -fL# -o /tmp/sysupgrade/sysupgrade-stage2 "https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/overlay/lower/usr/sbin/sysupgrade-stage2"
################################################################################################################# 100.0%
chmod: sysupgrade-stage2: No such file or directory
Found working directory /tmp/sysupgrade
Upgrade from a local file
Switching to working directory /tmp/sysupgrade
/tmp/sysupgrade
Running updated script.
Upgrading from a provided file
Unknown file
```

themactep, 2024-10-31:
please test and confirm that it works now

nicolasroche898, 2024-11-01:
tested, confirmed working with local path now, closing issue.

closed 2024-11-01 by nicolasroche898