themactep/thingino-firmware #288: U-boot hangs on Xiaomi MJSXJ03HL T31ZL jxq03p

Andrik45719, 2024-10-27:
Prebuild firmware for Xiaomi MJSXJ03HL hangs on u-boot stage.

```
Ingenic U-Boot Flex 2013.07 (Oct 13 2024 - 17:27:30)

Platform: ISVP (Ingenic XBurst1)
Built profile: T31L
SOC Name: T31ZL
DRAM:  64 MiB
Top of RAM usable for U-Boot at: 84000000
Reserving 471k for U-Boot at: 83f88000
Reserving 32832k for malloc() at: 81f78000
Reserving 32 Bytes for Board Info at: 81f77fe0
Reserving 124 Bytes for Global Data at: 81f77f64
Reserving 256k for boot params() at: 81f37f64
Stack Pointer at: 81f37f48
Now running in RAM - U-Boot at: 83f88000

MMC:   msc: 0
JZ SFC: Flash chip ID: 5e4018
SF:    Detected ZB25VQ128 (5e 40 18 5e ca)

```

Solved by replacing U-Boot binary by [u-boot-isvp_t31_sfcnor_lite.bin](https://github.com/gtxaspec/u-boot-ingenic/releases/download/latest/u-boot-isvp_t31_sfcnor_lite.bin)

```
Ingenic U-Boot Flex T31L SPL 2013.07-g00945d0 (Oct 27 2024 - 07:03:39)
Timer init
CLK stop
PLL init
pll_init:363
pll_cfg.pdiv = 10, pll_cfg.h2div = 5, pll_cfg.h0div = 5, pll_cfg.cdiv = 1, pll_cfg.l2div = 2
nf = 84 nr = 1 od0 = 1 od1 = 2
cppcr is 05405100
CPM_CPAPCR 0540510d
nf = 84 nr = 1 od0 = 1 od1 = 2
cppcr is 05405100
CPM_CPMPCR 07d0590d
nf = 100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPVPCR 0640510d
cppcr 0x9a773310
apll_freq = 1008000000
mpll_freq = 1000000000
vpll_freq = 1200000000
cpu clk source: apll
ddr clk source: mpll
ddrfreq 500000000
cclk  1008000000
l2clk 504000000
h0clk 200000000
h2clk 200000000
pclk  100000000
CLK init
SDRAM init
sdram init start
DDR clk rate 500000000
SDRAM init ok
board_init_r
image entry point: 0x80100000

Ingenic U-Boot Flex 2013.07-g00945d0 (Oct 27 2024 - 07:03:39)

Platform: ISVP (Ingenic XBurst1)
Built profile: T31L
SOC Name: T31ZL
DRAM:  64 MiB
Top of RAM usable for U-Boot at: 84000000
Reserving 702k for U-Boot at: 83f50000
Reserving 32832k for malloc() at: 81f40000
Reserving 32 Bytes for Board Info at: 81f3ffe0
Reserving 124 Bytes for Global Data at: 81f3ff64
Reserving 256k for boot params() at: 81efff64
Stack Pointer at: 81efff48
Now running in RAM - U-Boot at: 83f50000

MMC:   msc: 0
JZ SFC: Flash chip ID: 5e4018
SF:    Detected ZB25VQ128 (5e 40 18 5e c9)
Net:   HW address for ETH: 
Net:   HW address for WLAN: 
In:    serial
Out:   serial
Err:   serial
GPIO:  gpio_button: 51i
GPIO:  gpio_default: 25id 26iud
Net:   Network disabled
GPIO:  gpio_default_net: 39o
GPIO:  gpio_mmc_power: No GPIO env settings provided
GPIO:  gpio_ircut: 50i 49iud
GPIO:  gpio_user: No GPIO env settings provided
GPIO:  gpio_motor_v: No GPIO env settings provided
GPIO:  gpio_motor_h: No GPIO env settings provided
MMC:   Checking for autoupdate files...
MMC:   Card did not respond to voltage select!
MMC:   Checking for kernel image from MMC...
reading thingino_t31_kernel
MMC:   Kernel file not found.
MMC:   Checking for boot/env files...
MMC:   Card did not respond to voltage select!
** Bad device mmc 0 **
MMC:   Card did not respond to voltage select!
** Bad device mmc 0 **
KEY:   ###### Press Ctrl-C now to interrupt boot... loading in 3s ######
JZ SFC: Flash chip ID: 5e4018
SF:    Detected ZB25VQ128 (5e 40 18 5e 38)
SF:    probe command completed in 6ms
SQ:    Kernel start detected at address 0x50000
SQ:    SquashFS start detected at address 0x1B0000
SQ:    Overlay start set to address 0x7A0000
SQ:    SquashFS processing complete.
SF:    2424832 bytes @ 0x50000 Read: OK
SF:    read command completed in 779ms
## Booting kernel from Legacy Image at 0x80600000 ...
   Image Name:   Linux-3.10.14__isvp_swan_1.0__
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1391297 Bytes = 1.3 MiB
   Load Address: 80010000
   Entry Point:  8035e050
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

```

gtxaspec, 2024-10-31:
hmm... looks like we currently use `isvp_t31_xiaomi_sfcnor` which I presume is based on `isvp_t31_sfcnor`  
We probably need another xiaomi spl profile based on `isvp_t31_sfcnor_lite` to fix this...

gtxaspec, 2024-10-31:
Actually, `isvp_t31_xiaomi_sfcnor` is actually based on the lite version...
`isvp_t31_xiaomi_sfcnor      mips   xburst   isvp_t31    ingenic   t31   isvp_t31:SPL_SFC_SUPPORT,ENV_IS_IN_SPI_FLASH,SPL_SFC_NOR,JZ_MMC_MSC0,LITE_VERSION,XIAOMI_SPL`

@themactep anything I'm missing?

Andrik45719, 2024-10-31:
[xiaomi_mjsxj03hl_t31zl_uboot.bin.zip](https://github.com/user-attachments/files/17590175/xiaomi_mjsxj03hl_t31zl_uboot.bin.zip)
Stock u-boot dump
```
U-Boot SPL 2013.07-g8581847-dirty (Aug 09 2021 - 18:07:12)
Timer init
CLK stop
PLL init
pll_init:366
pll_cfg.pdiv = 10, pll_cfg.h2div = 5, pll_cfg.h0div = 5, pll_cfg.cdiv = 1, pll_cfg.l2div = 2
nf=84 nr = 1 od0 = 1 od1 = 2
cppcr is 05405100
CPM_CPAPCR 0540510d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPMPCR 0640510d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPVPCR 0640510d
cppcr 0x9a7b5510
apll_freq 1008000000
mpll_freq 1200000000
vpll_freq = 1200000000
ddr sel mpll, cpu sel apll
ddrfreq 600000000
cclk  1008000000
l2clk 504000000
h0clk 240000000
h2clk 240000000
pclk  120000000
CLK init
SDRAM init
sdram init start
ddr_inno_phy_init ..!
phy reg = 0x00000007, CL = 0x00000007
ddr_inno_phy_init ..! 11:  00000004
ddr_inno_phy_init ..! 22:  00000006
ddr_inno_phy_init ..! 33:  00000006
REG_DDR_LMR: 00000210
REG_DDR_LMR: 00000310
REG_DDR_LMR: 00000110
REG_DDR_LMR, MR0: 00f73011
T31_0x5: 00000007
T31_0x15: 0000000c
T31_0x4: 00000000
T31_0x14: 00000002
INNO_TRAINING_CTRL 1: 00000000
INNO_TRAINING_CTRL 2: 000000a1
T31_cc: 00000003
INNO_TRAINING_CTRL 3: 000000a0
T31_118: 0000003c
T31_158: 0000003c
T31_190: 0000001e
T31_194: 0000001c
jz-04 :  0x00000051
jz-08 :  0x000000a0
jz-28 :  0x00000024
DDR PHY init OK
INNO_DQ_WIDTH   :00000003
INNO_PLL_FBDIV  :00000014
INNO_PLL_PDIV   :00000005
INNO_MEM_CFG    :00000051
INNO_PLL_CTRL   :00000018
INNO_CHANNEL_EN :0000000d
INNO_CWL        :00000006
INNO_CL         :00000007
DDR Controller init
DDRC_STATUS         0x80000001
DDRC_CFG            0x0a288a40
DDRC_CTRL           0x0000011c
DDRC_LMR            0x00400008
DDRC_DLP            0x00000000
DDRC_TIMING1        0x050f0a06
DDRC_TIMING2        0x021c0807
DDRC_TIMING3        0x20080723
DDRC_TIMING4        0x1f240031
DDRC_TIMING5        0xff060405
DDRC_TIMING6        0x321c0505
DDRC_REFCNT         0x00917903
DDRC_MMAP0          0x000020fc
DDRC_MMAP1          0x00002400
DDRC_REMAP1         0x03020d0c
DDRC_REMAP2         0x07060504
DDRC_REMAP3         0x0b0a0908
DDRC_REMAP4         0x0f0e0100
DDRC_REMAP5         0x13121110
DDRC_AUTOSR_EN      0x00000000
sdram init finished
SDRAM init ok
board_init_r
image entry point: 0x80100000

U-Boot 2013.07-g8581847-dirty (Aug 09 2021 - 18:07:12)

Board: ISVP (Ingenic XBurst T31 SoC)
DRAM:  64 MiB
Top of RAM usable for U-Boot at: 84000000
Reserving 436k for U-Boot at: 83f90000
Reserving 32772k for malloc() at: 81f8f000
Reserving 32 Bytes for Board Info at: 81f8efe0
Reserving 124 Bytes for Global Data at: 81f8ef64
Reserving 128k for boot params() at: 81f6ef64
Stack Pointer at: 81f6ef48
Now running in RAM - U-Boot at: 83f90000
MMC:   msc: 0
the manufacturer 5e
SF: Detected ZB25VQ128

*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
misc_init_r before change the wifi_enable_gpio
gpio_request lable = wifi_enable_gpio gpio = 58
misc_init_r after gpio_request the wifi_enable_gpio ret is 58
misc_init_r after change the wifi_enable_gpio ret is 0
misc_init_r before change the yellow_gpio
gpio_request lable = yellow_gpio gpio = 39
misc_init_r after gpio_request the yellow_gpio ret is 39
misc_init_r after change the yellow_gpio ret is 1
misc_init_r before change the blue_gpio
gpio_request lable = blue_gpio gpio = 38
misc_init_r after gpio_request the blue_gpio ret is 38
misc_init_r after change the blue_gpio ret is 0
gpio_request lable = night_gpio gpio = 60
misc_init_r after gpio_request the night_gpio ret is 60
misc_init_r after change the night_gpio ret is 0
gpio_request lable = SPK_able_gpio gpio = 63
misc_init_r after gpio_request the SPK_able_gpio ret is 63
misc_init_r after change the SPK_able_gpio ret is 0
gpio_request lable = TF_en_gpio gpio = 47
misc_init_r after gpio_request the TF_en_gpio ret is 47
misc_init_r after change the TF_en_gpio ret is 0
gpio_request lable = TF_cd_gpio gpio = 48
misc_init_r after gpio_request the TF_cd_gpio ret is 48
misc_init_r after change the TF_cd_gpio ret is 1
gpio_request lable = SD_able_gpio gpio = 54
misc_init_r after change the SD_able_gpio ret is 0
misc_init_r before change the wifi_enable_gpio
gpio_request lable = wifi_enable_gpio gpio = 58
misc_init_r after gpio_request the wifi_enable_gpio ret is 58
misc_init_r after change the wifi_enable_gpio ret is 1
Hit any key to stop autoboot:  0
Card did not respond to voltage select!
SD card is not insert
gpio_request lable = sdupgrade gpio = 51
the manufacturer 5e
SF: Detected ZB25VQ128

The upgrade flag could not be found!
the manufacturer 5e
SF: Detected ZB25VQ128

--->probe spend 4 ms
SF: 2031616 bytes @ 0x40000 Read: OK
--->read spend 654 ms
## Booting kernel from Legacy Image at 80600000 ...

```

themactep, 2024-10-31:
It does not look like a uboot issue. Bootloader is all done at the moment and is passing the baton to kernel which either fails to start or has console disabled. Is it a regular thingino build or some frankenpatch?

gtxaspec, 2024-10-31:
from his log (the first one) it stalls right after detecting the flash chip.

themactep, 2024-10-31:
did not notice the first log on mobile.

Andrik45719, 2024-10-31:
As I see by defining XIAOMI_SPL Xiaomi bootloader just copied from firmware folder.

```
ifdef CONFIG_XIAOMI_SPL
ifdef CONFIG_SPL_SFC_SUPPORT
				cp $(TOPDIR)/firmware/t31_xiaomi_sfcnor.bin $(obj)spl/u-boot-spl-pad.bin
else
				$(error "T31_XIOAMI unsupported option")
endif
endif
		cat $(obj)spl/u-boot-spl-pad.bin $(obj)u-boot.bin > $@
		rm $(obj)spl/u-boot-spl-pad.bin

```
I am compared stub t31_xiaomi_sfcnor.bin and bootloader in thingino-xiaomi_mjsxj03hl_t31n_jxq03p.bin no diffs.

themactep, 2024-10-31:
that looks like a flash chip issue. it is either not supported in uboot, or has a wrong parameter. zbit chips have notoriosly bad support across ingenic ancient sources. even for cloner we had to add it manually. i have added them in our patched uboot but maybe the borrowed spl fucks it up. i would be ineterested to see how it works on the same camera with a different flash chip.

themactep, 2024-10-31:
i checked my 03hl. two t31n (one liberated, one with secureboot), one t31l with sb (not zl).

themactep, 2024-10-31:
[image]

But they use the same chip though. Never had an issue. Let me upgrade this one.

themactep, 2024-10-31:
```
Ingenic U-Boot Flex 2013.07 (Oct 29 2024 - 22:59:49)

Platform: ISVP (Ingenic XBurst1)
Built profile: T31L
SOC Name: Unknown
DRAM:  64 MiB
Top of RAM usable for U-Boot at: 84000000
Reserving 471k for U-Boot at: 83f88000
Reserving 32832k for malloc() at: 81f78000
Reserving 32 Bytes for Board Info at: 81f77fe0
Reserving 124 Bytes for Global Data at: 81f77f64
Reserving 256k for boot params() at: 81f37f64
Stack Pointer at: 81f37f48
Now running in RAM - U-Boot at: 83f88000

MMC:   msc: 0
JZ SFC: Flash chip ID: 5e4018
SF:    Detected ZB25VQ128 (5e 40 18 5e 00)
ENV:   Warning - bad CRC, using default environment
ENV:   Saving default environment...
ENV:   Erasing SPI flash...
ENV:   Writing to SPI flash... done
Net:   MAC address for ETH based on device serial set
JZ SFC: Flash chip ID: 5e4018
```

boom! hangs...

themactep, 2024-10-31:
> SOC Name: Unknown

themactep, 2024-10-31:
i think 32k vs 64k hit us again
https://github.com/gtxaspec/u-boot-ingenic/commit/8b543358666db8a029ed5267e71110bb2b522062

gtxaspec, 2024-10-31:
this wasn't an issue before? we went back to 64k.

themactep, 2024-10-31:
yes, in kernel, but as i see it is still 0x8000 or 32k in uboot.

gtxaspec, 2024-10-31:
ah right, i'll switch it back

Andrik45719, 2024-10-31:
Compiled u-boot (isvp_t31_sfcnor_lite) work fine on my T31ZL.

This one hangs (blob, secure boot?):
https://github.com/gtxaspec/u-boot-ingenic/blob/master/firmware/t31_xiaomi_sfcnor.bin

themactep, 2024-10-31:
Do you have a stock uboot from that camera? Change an insignificant piece of text in the file using a binhex editor ("SPL" to "SPY" or something) and flash it back. If it loads - no secure book. It it does not - secure boot.

gtxaspec, 2024-10-31:
it looks like his unit works with the normal one, and it doesn't work with the uboot that has the secureboot spl.

Andrik45719, 2024-10-31:
https://github.com/user-attachments/files/17590175/xiaomi_mjsxj03hl_t31zl_uboot.bin.zip

themactep, 2024-10-31:
mine that hangs is secure boot with t31n.

themactep, 2024-11-04:
Confirmed, the last commit in our uboot repo while seems completely unrelated breaks uboot builds with spliced foreign spl. @gtxaspec is going to fix it soon.

gtxaspec, 2024-11-04:
u-boot has been updated with some fixes, and the github CI is running, you can try to sysupgrade in about an hour and see if it works now.

Andrik45719, 2024-11-05:
Thanks!

```
stage 2 PATH is /tmp/sysupgrade
Running Stage 2 of sysupgrade.
Flashing firmware to /dev/mtd6
Erasing block: 512/512 (100%)
Writing kb: 16384/16384 (100%)
Verifying kb: 16384/16384 (100%)
Rebooting in 5 seconds...
```

```
U-Boot SPL 2013.07-g8581847-dirty (Aug 09 2021 - 18:07:12)
Timer init
CLK stop
PLL init
pll_init:366
pll_cfg.pdiv = 10, pll_cfg.h2div = 5, pll_cfg.h0div = 5, pll_cfg.cdiv = 1, pll_cfg.l2div = 2
nf=84 nr = 1 od0 = 1 od1 = 2
cppcr is 05405100
CPM_CPAPCR 0540510d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPMPCR 0640510d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPVPCR 0640510d
cppcr 0x9a7b5510
apll_freq 1008000000
mpll_freq 1200000000
vpll_freq = 1200000000
ddr sel mpll, cpu sel apll
ddrfreq 600000000
cclk  1008000000
l2clk 504000000
h0clk 240000000
h2clk 240000000
pclk  120000000
CLK init
SDRAM init
sdram init start
ddr_inno_phy_init ..!
phy reg = 0x00000007, CL = 0x00000007
ddr_inno_phy_init ..! 11:  00000004
ddr_inno_phy_init ..! 22:  00000006
ddr_inno_phy_init ..! 33:  00000006
REG_DDR_LMR: 00000210
REG_DDR_LMR: 00000310
REG_DDR_LMR: 00000110
REG_DDR_LMR, MR0: 00f73011
T31_0x5: 00000007
T31_0x15: 0000000c
T31_0x4: 00000000
T31_0x14: 00000002
INNO_TRAINING_CTRL 1: 00000000
INNO_TRAINING_CTRL 2: 000000a1
T31_cc: 00000003
INNO_TRAINING_CTRL 3: 000000a0
T31_118: 0000003c
T31_158: 0000003c
T31_190: 0000001e
T31_194: 0000001c
jz-04 :  0x00000051
jz-08 :  0x000000a0
jz-28 :  0x00000024
DDR PHY init OK
INNO_DQ_WIDTH   :00000003
INNO_PLL_FBDIV  :00000014
INNO_PLL_PDIV   :00000005
INNO_MEM_CFG    :00000051
INNO_PLL_CTRL   :00000018
INNO_CHANNEL_EN :0000000d
INNO_CWL        :00000006
INNO_CL         :00000007
DDR Controller init
DDRC_STATUS         0x80000001
DDRC_CFG            0x0a288a40
DDRC_CTRL           0x0000011c
DDRC_LMR            0x00400008
DDRC_DLP            0x00000000
DDRC_TIMING1        0x050f0a06
DDRC_TIMING2        0x021c0807
DDRC_TIMING3        0x20080723
DDRC_TIMING4        0x1f240031
DDRC_TIMING5        0xff060405
DDRC_TIMING6        0x321c0505
DDRC_REFCNT         0x00918c03
DDRC_MMAP0          0x000020fc
DDRC_MMAP1          0x00002400
DDRC_REMAP1         0x03020d0c
DDRC_REMAP2         0x07060504
DDRC_REMAP3         0x0b0a0908
DDRC_REMAP4         0x0f0e0100
DDRC_REMAP5         0x13121110
DDRC_AUTOSR_EN      0x00000000
sdram init finished
SDRAM init ok
board_init_r
image entry point: 0x80100000

Ingenic U-Boot Flex 2013.07 (Nov 01 2024 - 01:17:03)

Platform: ISVP (Ingenic XBurst1)
Built profile: T31L
SOC Name: T31ZL
DRAM:  64 MiB
Top of RAM usable for U-Boot at: 84000000
Reserving 471k for U-Boot at: 83f88000
Reserving 32832k for malloc() at: 81f78000
Reserving 32 Bytes for Board Info at: 81f77fe0
Reserving 124 Bytes for Global Data at: 81f77f64
Reserving 256k for boot params() at: 81f37f64
Stack Pointer at: 81f37f48
Now running in RAM - U-Boot at: 83f88000

MMC:   msc: 0
JZ SFC: Flash chip ID: 5e4018
SF:    Detected ZB25VQ128 (5e 40 18 5e 62)
ENV:   Warning - bad CRC, using default environment
ENV:   Saving default environment...
ENV:   Erasing SPI flash...
ENV:   Writing to SPI flash... done
Net:   MAC address for ETH based on device serial set
JZ SFC: Flash chip ID: 5e4018
SF:    Detected ZB25VQ128 (5e 40 18 5e 00)
ENV:   Erasing SPI flash...
ENV:   Writing to SPI flash... done
Net:   MAC address for WLAN based on device serial set
ENV:   Erasing SPI flash...
ENV:   Writing to SPI flash... done
In:    serial
Out:   serial
Err:   serial
KEY:   Reset button undefined
GPIO:  gpio_default: 25id 26iud
ETH:   Searching for valid PHY
ETH:   Invalid PHY found with ID 0:0x0-0x0
ETH:   Error: Invalid PHY configuration.
ETH:   PHY not found!
GPIO:  gpio_default_net: 39o
GPIO:  gpio_mmc_power: No GPIO env settings provided
GPIO:  gpio_ircut: 52i 53i 49i 50i 57i 58i
GPIO:  gpio_user: No GPIO env settings provided
GPIO:  gpio_motor_v: No GPIO env settings provided
GPIO:  gpio_motor_h: No GPIO env settings provided
MMC:   Checking for autoupdate files...
MMC:   Card did not respond to voltage select!
MMC:   Checking for kernel image from MMC...
reading thingino_t31_kernel
reading thingino_t31_kernel_1
reading thingino_t31_kernel_2
MMC:   Kernel file not found.
MMC:   Checking for boot/env files...
MMC:   Card did not respond to voltage select!
** Bad device mmc 0 **
MMC:   Card did not respond to voltage select!
** Bad device mmc 0 **
KEY:   ###### Press Ctrl-C now to interrupt boot... loading in 3s ######
JZ SFC: Flash chip ID: 5e4018
SF:    Detected ZB25VQ128 (5e 40 18 5e 00)
SF:    probe command completed in 6ms
SQ:    Kernel start detected at address 0x50000
SQ:    SquashFS start detected at address 0x1B0000
SQ:    Overlay start set to address 0x7A0000
SQ:    SquashFS processing complete.
SF:    2424832 bytes @ 0x50000 Read: OK
SF:    read command completed in 651ms
## Booting kernel from Legacy Image at 0x80600000 ...
   Image Name:   Linux-3.10.14__isvp_swan_1.0__
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1391177 Bytes = 1.3 MiB
   Load Address: 80010000
   Entry Point:  8035e050
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

```

closed 2024-11-05 by Andrik45719