themactep/thingino-firmware #290: Enhancement Request : "Privacy" Mode

timkeefer1, 2024-10-29:
I have a Home Assistant automation that - when I am home - turns off a smart switch that powers one of my cameras.  Frigate hates it - I get tons of watchdog, ffmpeg, and video ERROR message in my logs as it keeps retrying the camera.  I'm wondering if a pseudo "privacy mode" could be created for Thingino that could return a blank black camera feed.  Basically creating a software "shutter" for the camera?

timkeefer1, 2024-10-29:
> Sure. You can use "logo" as a privacy mask. Create an image that fills all of your sensor's output resolution, place it at 0,0 coordinates and there you go. https://github.com/gtxaspec/prudynt-t/blob/master/prudynt.cfg.example

That looks like it would work!  Now I just need to be able to curl -> shell script to make the cfg changes and reload... is there API docs for prudynt?

tERB00, 2024-10-29:
There is no API but you can inspect what does your browser sending to the camera when you tick "Logo" checkbox on the OSD settings page. Should not be anything too special I reckon.

themactep, 2025-12-29:
implemented