themactep/thingino-firmware #291: sysupgrade 'Upgrade from a local file' not work

Andrik45719, 2024-10-29:
ssh
```
 sysupgrade /mnt/mmcblk0p1/thingino-xiaomi_mjsxj03hl_t31l_jxq03p-update.bin
/sbin/sysupgrade: line 168: echo_d: not found
Upgrade from a local file
```
serial console
```
sysupgrade /mnt/mmcblk0p1/thingino-xiaomi_mjsxj03hl_t31n_jxq03p-update.bin
Run as /sbin/sysupgrade /mnt/mmcblk0p1/thingino-xiaomi_mjsxj03hl_t31n_jxq03p-update.bin
Creating /tmp/sysupgrade directory
Upgrade from a local file
Found /mnt/mmcblk0p1/thingino-xiaomi_mjsxj03hl_t31n_jxq03p-update.bin

Ingenic U-Boot Flex T31L SPL 2013.07-g00945d09-dirty (Oct 28 2024 - 15:58:12)
```
While manual firmware update via flashcp <file> /dev/mtd5 succeed

themactep, 2024-10-31:
how old is your build?

Andrik45719, 2024-10-31:
Thingino xiaomi_mjsxj03hl_t31n_jxq03p [master+16990ed, 2024-10-27 04:15:46 -0400]

nicolasroche898, 2024-11-01:
from wiki:

Cameras like Xiaomi MJSXJ03HL with low-end SoC, large sensor resolution, and large flash memory (16MB) can have issues trying to install an update:

```
   \\   _______ _     _ _____ __   _  ______ _____ __   _  _____
   )\\     |    |_____|   |   | \  | |  ____   |   | \  | |     |
  (  /     |    |     | __|__ |  \_| |_____| __|__ |  \_| |_____|
  / /
        xiaomi_mjsxj03hl_t31n_jxq03p
        master+8b67c2e, 2024-10-02 00:02:24 -0400

root@ing-xiaomi-mjsxj03hl-8a0a ~# sysupgrade -p
Partial upgrade from GitHub
Upgrading from GitHub
Not enough free space to download firmware: 16449536 > 13631488
```

To upgrade such a camera from Linux, you first need to change the memory mapping to give the system more space:

```
fw_setenv osmem 52M@0x0
fw_setenv rmem 12M@0x3400000
reboot
```

Then do the upgrade and reset the memory mapping back:

```
fw_setenv osmem 32M@0x0
fw_setenv rmem 32M@0x2000000
reboot
```

Andrik45719, 2024-11-01:
I am install from mmc card not from ram drive.

themactep, 2024-11-01:
Download a fresh version of sysupgrade on the camera and run it. The script will upgrade itself first before performing a system upgrade.

Installing from a local file still requires enough space in tmpfs as the local file gets copied into the working directory for the second stage of the installation.

themactep, 2024-11-08:
it that still an issue?

closed 2024-11-10 by themactep