themactep/thingino-firmware #296: Settings (webui) do not load through Reverse Proxy

timkeefer1, 2024-10-30:
I'm running nginx proxy manager and passing the web admin interface (using SSL/CF tunnel).   When I try and pull up the existing settings through the webui, none of the configuration settings populate.  If I hit the IP of the device directly, they load in fine.

nicolasroche898, 2024-11-01:
it sounds like your proxy isn't passing websocket just http. settings are passed via websocket.

nicolasroche898, 2024-11-01:
aha websockets are on port 8089, you need to pass this via your proxy too.

config-streamer.cgi:
new WebSocket('ws://' + document.location.hostname + ':8089?token=

timkeefer1, 2024-11-02:
OK so after 36 versions of NPM advanced configuration settings I've finally got it... here's a version that should work for anyone that is using a similar setup to me.  Adding just in case this helps folks who may stumble across this in the future!

```
# NPM (NGINX Proxy Manager) Configuration for Thingino IP Cameras with WebSocket Support
# This configuration enables secure streaming over HTTPS/WSS

location / {
    # Authentication settings
    auth_basic off;
    # Replace the text after "Basic" with your own base64 encoded "username:password"
    # To generate: echo -n "your_username:your_password" | base64
    proxy_set_header Authorization "Basic YOUR_BASE64_AUTH_STRING";
        
    # Should pull in the correct variable from $server but you can replace instead
    #   Replace $server with your camera's IP address (e.g., 192.168.1.100)
    proxy_pass http://$server;
    
    # Required headers - do not modify
    proxy_set_header Host $host;
    proxy_set_header X-Real-IP $remote_addr;
    proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
    proxy_set_header X-Forwarded-Proto $scheme;
    
    # WebSocket support - required for streaming
    proxy_http_version 1.1;
    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection "upgrade";
    
    # Stream settings - recommended to keep as is
    proxy_buffering off;
    
    # WebSocket URL conversion for secure streaming
    # These replacements handle the ws:// to wss:// conversion
    sub_filter "ws://" "wss://";
    sub_filter ":8089" "";
    sub_filter_once off;
    
    # WebSocket connection handler
    # This routes WebSocket traffic to port 8089 on your camera
    # Should pull in the correct variable from $server but you can replace instead
    #   Replace $server with your camera's IP address (e.g., 192.168.1.100)
    if ($query_string ~* "token=") {
        proxy_pass http://$server:8089;
    }
}

# Required NPM Settings (enable these in the UI):
# 1. SSL: Enabled
# 2. WebSocket Support: Enabled
# 3. Block Common Exploits: Disabled (if you experience issues)

# Usage Instructions:
# 1. Replace YOUR_BASE64_AUTH_STRING with your base64 encoded credentials
# 2. Paste this configuration in the "Custom Configuration" field under
#    the Advanced tab in your NPM proxy host settings
```

themactep, 2024-11-08:
fixed. please test and confirm.

closed 2024-11-10 by themactep