themactep/thingino-firmware #300: bug: webui: reset firmware doesn't work

nicolasroche898, 2024-11-01:
repro:
1. in web ui go to: settings -> reset
2. click: reset firmware button + Yes

expected:
firmware is reset.

actual:
this is printed for 1 second on the web page, then goes to "restarting" page, but camera is not restarted and continues as normal (not reset)...

```
Erase overlay
[@ing-jooan-a6m-654a /tmp]# firstboot

!!! Warning !!!

This action will reset the device to defaults and erase all configurations, leaving the device like the first boot.

All saved settings will be erased!

Are you sure you want to proceed? (yes/no)
Operation aborted.

EXIT CODE: 1

```

themactep, 2024-11-04:
please test with the new build and see it it works now

nicolasroche898, 2024-11-04:
new test on:
BUILD_ID="master+a4ca4b2, 2024-11-04 09:36:06 -0500"

steps:
wipe flash and install new firmware
change and save any setting
go to: settings -> reset -> reset firmware -> Yes

result:
camera is reset to original state correctly and transmitting THINGINO AP.
bug fixed.

web browser stays in this state:
i guess there isn't much that can be done about that, as camera won't be accessible anymore after reset.

[image]

themactep, 2024-11-07:
i replaced the buttons with a link to a wiki article about camera configuration.

themactep, 2024-11-08:
fixed.

closed 2024-11-08 by themactep