themactep/thingino-firmware #302: use protocol-relative websocket on all cgi pages

nicolasroche898, 2024-11-02:
in all cgi pages, convert this:
`let ws = new WebSocket('ws://' + document.location.hostname + ':8089?token=<%= $ws_token %>');`
into this:
`let ws = new WebSocket('//' + document.location.hostname + ':8089?token=<%= $ws_token %>');`
this way the ui will run both for http (ws://) and for https (wss://) (via proxy) without any rewrite rules.

[related](https://github.com/themactep/thingino-firmware/issues/296#issuecomment-2452989238)

themactep, 2024-11-08:
done. please test and confirm it's solved.

nicolasroche898, 2024-11-10:
tested, confirmed solved.

closed 2024-11-10 by nicolasroche898

jspiros, 2025-05-31:
FYI, this change broke the Thingino web interface for users of Firefox ESR 115.x, as support for WebSocket URLs that lack the protocol component wasn't added until Firefox 124 (https://bugzilla.mozilla.org/show_bug.cgi?id=1797449). Under Firefox 115.x, the changed lines cause a DOMException and the page is non-functional. As of now, ESR 115 remains the supported version of Firefox for Windows 7-8.1 and macOS 10.12-10.14.

> For folks affected by the change in browser behavior, here's the status as of today (2024-05-10):
> 
> 
> 
> - Firefox shipped in 124.0 (released 2024-03-19)
> 
> - Safari shipped this in 17 a.k.a. 19616.1.27 (released 2023-09-18)
> 
> - Chromium shipped this in 125 (released 2024-05-08)
> 
>   - ticket: https://issues.chromium.org/issues/40244242
> 
>   - review: https://chromium-review.googlesource.com/c/chromium/src/+/5060558
> 
>   - roadmap: https://chromestatus.com/feature/5848709993857024
> 
> 
> 
> The previously-mentioned https://github.com/whatwg/websockets/pull/45 has relevant links to track the implementations. 

 _Originally posted by @qguv in [#20](https://github.com/whatwg/websockets/issues/20#issuecomment-2011908591)_