themactep/thingino-firmware #306: display a static image in preview if streamer is not running

gtxaspec, 2024-11-04:
signal to user streamer is offline.

themactep, 2024-11-07:
a static image of what exactly? a dead streamer?

gtxaspec, 2024-11-08:
this is what we see right now if the streamer is down:

[image]

maybe an image of a camera offline-ish logo? something to indicate of the status...
would be nice to have the same one if the stream is disabled too.

gtxaspec, 2024-11-08:
kind of like (from chatgpt lol)
[image]

themactep, 2024-11-08:
We can do better :)

[image]

gtxaspec, 2024-11-08:
lol!!! epic.

gtxaspec, 2024-11-09:
added, thanks @themactep !

closed 2024-11-09 by gtxaspec