themactep/thingino-firmware #311: T31n HI3881 wlan problem

daoudeddy, 2024-11-06:
hello, i have an Ingenic T31n board with HI3881 wlan chip and i can't get it to work

dmesg 

```[VERSION]:Hi3881V100R001C00SPC021 2020-10-14 20:25:00
oam_main_init SUCCESSFULLY!
wal_customize_init SUCCESSFULLY!
oal_main_init SUCCESSFULLY
frw_main_init SUCCESSFULLY!
hi_wifi_plat_init SUCCESSFULLY
mmc1: host doesn't support card's voltages
mmc1: error -22 whilst initialising SDIO card
mmc1: host doesn't support card's voltages
mmc1: error -22 whilst initialising MMC card
mmc1: host doesn't support card's voltages
mmc1: error -22 whilst initialising SDIO card
mmc1: host doesn't support card's voltages
mmc1: error -22 whilst initialising MMC card
mmc1: host doesn't support card's voltages
mmc1: error -22 whilst initialising SDIO card
mmc1: host doesn't support card's voltages
mmc1: error -22 whilst initialising MMC card
[ERROR]:oal_sdio_host.c:oal_sdio_func_probe:2257:sdio enum timeout, reason[probe timeout]

[ERROR]:1646:hcc_host_init:: sdio_probe failed! 
CPU 0 Unable to handle kernel paging request at virtual address 0000000c, epc == 8033a084, ra == 8033a2e0
Oops[#1]:
CPU: 0 PID: 1180 Comm: modprobe Tainted: G           O 3.10.14__isvp_swan_1.0__ #1
task: 821e2760 ti: 81912000 task.ti: 81912000
$ 0   : 00000000 10001c00 00000000 fffffffe
$ 4   : 8187fab4 00000001 00000000 000000a2
$ 8   : 0000000d 000000a2 00000000 ffffffe0
$12   : 736f685f 80420000 00000000 6e695f74
$16   : 8187fab4 00000000 803d2970 827c703c
$20   : 00000001 c04d0000 803c0000 80348114
$24   : 00000001 80064158                  
$28   : 81912000 81913c20 80348134 8033a2e0
Hi    : 00000000
Lo    : 00000000
epc   : 8033a084 0x8033a084
    Tainted: G           O
ra    : 8033a2e0 0x8033a2e0
Status: 10001c03        KERNEL EXL IE 
Cause : 40808008
BadVA : 0000000c
PrId  : 00d00100 (Ingenic Xburst)
Modules linked in: hi3881(O+) sensor_sc2332_t31(O) pwm_hal(O) pwm_core(O) tx_isp_t31(O) avpu(O) exfat(O) motor(O) jzmmc_v12
Process modprobe (pid: 1180, threadinfo=81912000, task=821e2760, tls=77945da4)
Stack : 00000000 00000000 00000000 00000000 00000000 8187fab4 00000000 8033a2e0
          8187fa80 801126c8 00000000 00000000 803d8140 803d8140 8187fab4 821e2760
          00000000 c04cbca0 c04cbca0 801d02a8 00000000 00000000 00000000 00000000
          827c7000 c049b8a4 8033d158 c0417b14 827c7000 c049b8a4 8033d158 827c703c
          00000001 c0499b98 827c7000 803c0000 00000000 80038854 00000000 00000000
          ...
Call Trace:[<8033a2e0>] 0x8033a2e0
[<801126c8>] 0x801126c8
[<801d02a8>] 0x801d02a8
[<c049b8a4>] 0xc049b8a4
[<8033d158>] 0x8033d158
[<c0417b14>] 0xc0417b14
[<c049b8a4>] 0xc049b8a4
[<8033d158>] 0x8033d158
[<c0499b98>] 0xc0499b98
[<80038854>] 0x80038854
[<80010574>] 0x80010574
[<c0491fd8>] 0xc0491fd8
[<8033d158>] 0x8033d158
[<c049b8a4>] 0xc049b8a4
[<8033d190>] 0x8033d190
[<801948d0>] 0x801948d0
[<c049289c>] 0xc049289c
[<80010574>] 0x80010574
[<c049cdd4>] 0xc049cdd4
[<8033d158>] 0x8033d158
[<c0491fd8>] 0xc0491fd8
[<8033d158>] 0x8033d158
[<80010574>] 0x80010574
[<8033d158>] 0x8033d158
[<80010574>] 0x80010574
[<c049cdd4>] 0xc049cdd4
[<80192988>] 0x80192988
[<80341bd4>] 0x80341bd4
[<8005db7c>] 0x8005db7c
[<c049cbc8>] 0xc049cbc8
[<80010574>] 0x80010574
[<8005dbb4>] 0x8005dbb4
[<800782f0>] 0x800782f0
[<8007601c>] 0x8007601c
[<80078690>] 0x80078690
[<8002257c>] 0x8002257c

Code: 8c820000  2403fffe  00431024 <8c51000c> 8f820014  00808025  24420001  10a00004  af820014 
---[ end trace a87155b422f25980 ]---
Bus Mode Reg after reset: 0x00020101, cnt=0
```
Config

``` # FRAG: soc toolchain ccache brand rootfs kernel system target uboot
BR2_SOC_INGENIC_T31N=y
BR2_SENSOR_SC2332=y
BR2_PACKAGE_WIFI=y
BR2_PACKAGE_WIFI_HI3881=y
BR2_ETHERNET=y
BR2_PACKAGE_THINGINO_KOPT_JZ_MAC=y
BR2_PACKAGE_THINGINO_KOPT_JZ_MAC_V12=y
BR2_PWM_ENABLE=y
BR2_AUDIO=y
BR2_MOTORS=y
BR2_PACKAGE_THINGINO_KOPT_MMC0=y
BR2_PACKAGE_THINGINO_KOPT_MMC0_PB_4BIT=y
BR2_PACKAGE_WIFI=y
BR2_PACKAGE_WIFI_HI3881=y
BR2_SDCARD=y
FLASH_SIZE_8=y
 ```

Original firmware scripts

```configWifi.sh```
```
insmod /mnt/flash/Server/resource/wifi/hi3881.ko g_fw_mode=1

MACINFO=$(ifconfig wlan0|grep HWaddr|awk '{print $5}')
echo "Got random macInfo: ${MACINFO}"

ifconfig wlan0 up
ifconfig wlan0 down
echo "wlan0 al_tx 1 0 20 1 7" > /sys/hisys/hipriv
ifconfig wlan0 up
echo "wlan0 set_cal_freq -15" > /sys/hisys/hipriv
echo "wlan0 set_cal_bpwr 0 20" > /sys/hisys/hipriv
echo "wlan0 al_tx 0" > /sys/hisys/hipriv
ifconfig wlan0 down
echo "wlan0 al_tx 1 0 20 7 7" > /sys/hisys/hipriv
ifconfig wlan0 up
echo "wlan0 set_cal_bpwr 1 15" > /sys/hisys/hipriv
echo "wlan0 al_tx 0" > /sys/hisys/hipriv
ifconfig wlan0 down
echo "wlan0 al_tx 1 0 20 13 7" > /sys/hisys/hipriv
ifconfig wlan0 up
echo "wlan0 set_cal_bpwr 0 15" > /sys/hisys/hipriv
echo "wlan0 set_cal_bpwr 1 15" > /sys/hisys/hipriv
echo "wlan0 set_cal_bpwr 2 15" > /sys/hisys/hipriv
echo "wlan0 al_tx 0" > /sys/hisys/hipriv
ifconfig wlan0 down

echo "wlan0 w_cal_data 0" > /sys/hisys/hipriv
sleep 3
echo "wlan0 set_efuse_mac ${MACINFO} 0" > /sys/hisys/hipriv

sleep 3
reboot
```

``` autorun.sh ```
```lsusb | awk '{print $6}' | grep -v "1d6b" || echo > /tmp/NoneUsbdevInserted

if [ -e /mnt/flash/Server/resource/wifi/hi3881.ko ]; then
	rm -rf /tmp/NoneUsbdevInserted
	cd /sys/class/gpio/
	echo 6 > export
	echo out > gpio6/direction
	echo 1 > gpio6/value
	sleep 1
	echo 0 > gpio6/value 
	sleep 1
	echo 1 > gpio6/value
	cd -
	
	mkdir -p /tmp/system/lib/firmware
	mkdir -p /tmp/system/lib/modules 
	mkdir -p /tmp/system/lib/modules/3.10.14__isvp_swan_1.0__ 
	
	mount --bind /tmp/system/ /system/
	
	#mount --bind /mnt/flash/Server/etc/firmware/ /lib/firmware
fi
```

[Original kernel module hi3881.zip](https://github.com/user-attachments/files/17647948/hi3881.zip)

gtxaspec, 2024-11-06:
you need
```
BR2_PACKAGE_THINGINO_KOPT_MMC1=y
BR2_PACKAGE_THINGINO_KOPT_MMC1_PB_4BIT=y
```
in the config too.  try that.

daoudeddy, 2024-11-06:
> you need
> 
> ```
> BR2_PACKAGE_THINGINO_KOPT_MMC1=y
> BR2_PACKAGE_THINGINO_KOPT_MMC1_PB_4BIT=y
> ```
> 
> in the config too. try that.

tried that too with the same error

and also tried 
BR2_PACKAGE_THINGINO_KOPT_MMC1=y
BR2_PACKAGE_THINGINO_KOPT_MMC1_P -> C <- _4BIT=y

daoudeddy, 2024-11-06:
> the correct one is `PB`.
> 
> Did you specify the variable `wlan_gpio` ?
> 
> I think you need to properly create _module_, and _camera_ configs for your device, along with the _environment_ config.
> 
> see the existing examples in the repo for all three. this will make testing and setup easier.
> 
> module: https://github.com/themactep/thingino-firmware/blob/master/configs/modules/t31x_gc2053_rtl8189ftv camera: https://github.com/themactep/thingino-firmware/blob/master/configs/cameras/wyze_c3_t31x_rtl environment: https://github.com/themactep/thingino-firmware/blob/master/environment/wyze-c3-t31x-rtl.uenv.txt

no success after trying your suggestion

gtxaspec, 2024-11-06:
please share each of your config files.

daoudeddy, 2024-11-06:
> please share each of your config files.

config/camera
```
# MODULE: t31n_sc2332_hi3881
BR2_AUDIO=y
BR2_PACKAGE_THINGINO_KOPT_MMC0=y
BR2_PACKAGE_THINGINO_KOPT_MMC0_PB_4BIT=y
BR2_PACKAGE_THINGINO_KOPT_MMC1=y
BR2_PACKAGE_THINGINO_KOPT_MMC1_PB_4BIT=y
BR2_MOTORS=y
BR2_PWM_ENABLE=y
BR2_SDCARD=y
FLASH_SIZE_8=y
U_BOOT_ENV_TXT="$(BR2_EXTERNAL)/environment/hc-3301y-ht1.uenv.txt"
```

config/modules
```
# FRAG: soc toolchain ccache brand rootfs kernel system target uboot
BR2_SOC_INGENIC_T31N=y
BR2_SENSOR_SC2332=y
BR2_ETHERNET=y
BR2_PACKAGE_THINGINO_KOPT_JZ_MAC_V12=y
BR2_PACKAGE_WIFI=y
BR2_PACKAGE_WIFI_HI3881=y
```

env
```
day_night_max=15000
day_night_min=500
disable_eth=true
enable_updates=true
gpio_motor_v=53 52 54 51
gpio_motor_h=57 58 61 62
gpio_ircut=63 64
motor_speed=200
gpio_speaker=14
gpio_default=56 52 54 51 57 58 61 62 63 64 49 50
motor_maxstep_h=1000
motor_maxstep_v=1100
gpio_white=49o
gpio_ir850=50o
motor_speed_h=500
motor_speed_v=500
motor_pos_0=428,185
```

gtxaspec, 2024-11-06:
in module, change
`BR2_PACKAGE_THINGINO_KOPT_JZ_MAC_V12=y`
to
`BR2_PACKAGE_THINGINO_KOPT_JZ_MAC_V13=y` 
(v12 doesn't work on t31"

in env. add
`gpio_wlan=6O`
change
`gpio_default=56 52 54 51 57 58 61 62 63 64 49 50`
to
`gpio_default=6o 6O 56 52 54 51 57 58 61 62 63 64 49 50`

do a full rebuild:
`make distclean`
then
`make fast`

daoudeddy, 2024-11-06:
Still no sucess 

```
Ingenic on chip Ethernet MAC driver V13
wait stable.[289][cgu_macphy]
Bus Mode Reg after reset: 0x00020101, cnt=0
libphy: jz_mii_bus: probed
=======>gmacdev = 0x822a2400<================
=========>gmacdev->MacBase = 0xb34b0000 DmaBase = 0xb34b1000
Bus Mode Reg after reset: 0x00020101, cnt=0
jz_mac jz_mac.0: JZ on-chip Ethernet MAC driver, Version 1.0
i2c /dev entries driver
jz-wdt: watchdog initialized
TCP: cubic registered
NET: Registered protocol family 10
NET: Registered protocol family 17
input: gpio-keys as /devices/platform/gpio-keys/input/input0
VFS: Mounted root (squashfs filesystem) readonly on device 31:3.
devtmpfs: mounted
Freeing unused kernel memory: 212K (803db000 - 80410000)
jffs2: notice: (374) jffs2_build_xattr_subsystem: complete building xattr subsystem, 1 of xdatum (0 unchecked, 0 orphan) and 8 of xref (0 dead, 7 orphan) found.
exFAT: Version 1.2.9
request spk en gpio 14 ok!
jz_codec_register: probe() successful!
cgu_set_rate, parent = 1008000000, rate = 2048000, n = 7875, reg val = 0x01001ec3
cgu_enable,cgu_i2s_spk reg val = 0x21001ec3
cgu_set_rate, parent = 1008000000, rate = 2048000, n = 7875, reg val = 0x01001ec3
cgu_enable,cgu_i2s_mic reg val = 0x21001ec3
dma dma0chan24: Channel 24 have been requested.(phy id 7,type 0x06 desc a279a000)
dma dma0chan25: Channel 25 have been requested.(phy id 6,type 0x06 desc a27f9000)
dma dma0chan26: Channel 26 have been requested.(phy id 5,type 0x04 desc a279f000)
@@@@ avpu driver ok(version H20220825a) @@@@@
@@@@ tx-isp-probe ok(version H20221206a), compiler date=Dec  6 2022 @@@@@
jz_pwm_probe[359] d_name = tcu_chn0
jz_pwm_probe[359] d_name = tcu_chn1
jz_pwm_probe[359] d_name = tcu_chn2
jz_pwm_probe[359] d_name = tcu_chn3
The version of PWM driver is H20210412a
request pwm channel 0 successfully
request pwm channel 1 successfully
request pwm channel 2 successfully
request pwm channel 3 successfully
pwm-jz pwm-jz: jz_pwm_probe register ok !
jzmmc_v1.2 jzmmc_v1.2.0: vmmc regulator missing
jzmmc_v1.2 jzmmc_v1.2.0: card-detect gpio pin not specified, card-detect disabled.  Error: host->pdata->removal = 1, errno=-1
jzmmc_v1.2 jzmmc_v1.2.0: register success!
jzmmc_v1.2 jzmmc_v1.2.1: vmmc regulator missing
jzmmc_v1.2 jzmmc_v1.2.1: register success!
[VERSION]:Hi3881V100R001C00SPC021 2020-10-14 20:25:00
oam_main_init SUCCESSFULLY!
wal_customize_init SUCCESSFULLY!
oal_main_init SUCCESSFULLY
frw_main_init SUCCESSFULLY!
hi_wifi_plat_init SUCCESSFULLY
mmc1: host doesn't support card's voltages
mmc1: error -22 whilst initialising SDIO card
mmc1: host doesn't support card's voltages
mmc1: error -22 whilst initialising MMC card
mmc1: host doesn't support card's voltages
mmc1: error -22 whilst initialising SDIO card
mmc1: host doesn't support card's voltages
mmc1: error -22 whilst initialising MMC card
mmc1: host doesn't support card's voltages
mmc1: error -22 whilst initialising SDIO card
mmc1: host doesn't support card's voltages
mmc1: error -22 whilst initialising MMC card
[ERROR]:oal_sdio_host.c:oal_sdio_func_probe:2257:sdio enum timeout, reason[probe timeout]

[ERROR]:1646:hcc_host_init:: sdio_probe failed!
CPU 0 Unable to handle kernel paging request at virtual address 0000000c, epc == 8033a084, ra == 8033a2e0
Oops[#1]:
CPU: 0 PID: 1114 Comm: modprobe Tainted: G           O 3.10.14__isvp_swan_1.0__ #1
task: 821a93b0 ti: 819c0000 task.ti: 819c0000
$ 0   : 00000000 10001c00 00000000 fffffffe
$ 4   : 81da33b4 00000001 00000000 000000a5
$ 8   : 0000000d 000000a5 00000000 ffffffe0
$12   : 736f685f 80420000 00000000 6e695f74
$16   : 81da33b4 00000000 803d2970 826d703c
$20   : 00000001 c04f0000 803c0000 80348114
$24   : 00000001 80064158
$28   : 819c0000 819c1c20 80348134 8033a2e0
Hi    : 00000000
Lo    : 00000000
epc   : 8033a084 0x8033a084
    Tainted: G           O
ra    : 8033a2e0 0x8033a2e0
Status: 10001c03        KERNEL EXL IE
Cause : 40808008
BadVA : 0000000c
PrId  : 00d00100 (Ingenic Xburst)
Modules linked in: hi3881(O+) jzmmc_v12 sensor_sc2332_t31(O) pwm_hal(O) pwm_core(O) tx_isp_t31(O) avpu(O) audio(O) exfat(O)
Process modprobe (pid: 1114, threadinfo=819c0000, task=821a93b0, tls=7706eda4)
Stack : 00000000 00000000 00000000 00000000 00000000 81da33b4 00000000 8033a2e0
          81da3380 801126c8 00000000 00000000 803d8140 803d8140 81da33b4 821a93b0
          00000000 c04eaca0 c04eaca0 801d02a8 00000000 00000000 00000000 00000000
          826d7000 c04ba8a4 8033d158 c0436b14 826d7000 c04ba8a4 8033d158 826d703c
          00000001 c04b8b98 826d7000 803c0000 00000000 80038854 00000000 00000000
          ...
Call Trace:[<8033a2e0>] 0x8033a2e0
[<801126c8>] 0x801126c8
[<801d02a8>] 0x801d02a8
[<c04ba8a4>] 0xc04ba8a4
[<8033d158>] 0x8033d158
[<c0436b14>] 0xc0436b14
[<c04ba8a4>] 0xc04ba8a4
[<8033d158>] 0x8033d158
[<c04b8b98>] 0xc04b8b98
[<80038854>] 0x80038854
[<80010574>] 0x80010574
[<c04b0fd8>] 0xc04b0fd8
[<8033d158>] 0x8033d158
[<c04ba8a4>] 0xc04ba8a4
[<8033d190>] 0x8033d190
[<801948d0>] 0x801948d0
[<c04b189c>] 0xc04b189c
[<80010574>] 0x80010574
[<c04bbdd4>] 0xc04bbdd4
[<8033d158>] 0x8033d158
[<c04b0fd8>] 0xc04b0fd8
[<8033d158>] 0x8033d158
[<80010574>] 0x80010574
[<8033d158>] 0x8033d158
[<80010574>] 0x80010574
[<c04bbdd4>] 0xc04bbdd4
[<80192988>] 0x80192988
[<80341bd4>] 0x80341bd4
[<8005db7c>] 0x8005db7c
[<c04bbbc8>] 0xc04bbbc8
[<80010574>] 0x80010574
[<8005dbb4>] 0x8005dbb4
[<800782f0>] 0x800782f0
[<8007601c>] 0x8007601c
[<80078690>] 0x80078690
[<8002257c>] 0x8002257c

Code: 8c820000  2403fffe  00431024 <8c51000c> 8f820014  00808025  24420001  10a00004  af820014
---[ end trace e257d74b0a76422c ]---
```

gtxaspec, 2024-11-06:
Can you share thingino-diag -l from there?

daoudeddy, 2024-11-07:
> Can you share thingino-diag -l from there?
> 
> also, do you have the gpio map from the stock firmware? that would be helpful in confirming that gpio 6 is indeed for the wifi module.

Kernel debug was disabled on the stock firmware so i had to find all the gpios manually
and i could only find in the gpio/exports 
gpio 6 7 14 63 64

gtxaspec, 2024-11-07:
are you able to join the discord or telegram groups? would make debugging with easier.

daoudeddy, 2024-11-30:
changing the MMC config to PA fixes the issue

```BR2_PACKAGE_THINGINO_KOPT_MMC1=y```
```BR2_PACKAGE_THINGINO_KOPT_MMC1_PA_4BIT=y```

```[   89.519012] jzmmc_v1.2 jzmmc_v1.2.1: card insert via sysfs
[   89.708842] mmc1: queuing unknown CIS tuple 0x91 (5 bytes)
[   89.708896] mmc1: new SDIO card at address 0001
[   89.791493] [VERSION]:Hi3881V100R001C00SPC021 2020-10-14 20:25:00
[   89.793971] oam_main_init SUCCESSFULLY!
[   89.793993] wal_customize_init SUCCESSFULLY!
[   89.794027] oal_main_init SUCCESSFULLY
[   89.797103] frw_main_init SUCCESSFULLY!
[   89.797119] hi_wifi_plat_init SUCCESSFULLY
[   89.802737] hcc_hmac_init SUCCESSFULLY
[   89.802848] plat_firmware_init SUCCESSFULLY
[   90.034569] Device is Ready!
[   90.034699] wlan_pm_open SUCCESSFULLY!!
[   90.039337] hmac_main_init SUCCESSULLY
[   90.044792] firmware_sync_cfg_paras_to_wal_customize SUCCESSFULLY
[   90.061085] wal_main_init SUCCESSFULLY
[   90.061100] hi_wifi_host_init SUCCESSFULLY
[   90.061506] WiFi driver init SUCCESSFULLY!
```

gtxaspec, 2024-12-03:
looks like it works! :smile_cat:

closed 2024-12-03 by gtxaspec