themactep/thingino-firmware #317: AP mode enhancements

goatzillax, 2024-11-13:
I'd like to hook up multiple cameras to one that is acting as host in a remote environment.  This seems to work pretty well, but could use some tweaks.

AP mode should probably run the ntp server to serve time out to clients; this seems to already be built into BusyBox; just need to turn it on in AP mode.

udhcpd seems to be behaving strangely -- the pid file doesn't show up, and the lease file isn't updating.  If I give it USR1 it will update, but it seems to not be doing it on its own.

The AP should probably add its own configured hostname to dnsd-ap.conf.

WiFi client hostnames aren't showing up in dnsd-ap.conf as well; not sure if this should be handled with some script called by udhcp to update dnsd-ap.conf on new leases; maybe that file needs to be tossed into /tmp and generated as well.  Might need dumpleases turned on...

Ultimately it might be nice to switch in/out of AP mode either with the reset button or maybe some magic SD card, but baby steps...

Thanks!

gtxaspec, 2024-11-13:
The `dnsd` and `dhcpd` daemons in BusyBox are minimal but serve our current needs and have a small footprint, which is ideal for now.

Looking ahead, it would be beneficial to develop an "AP Self-Hosted Server Mode." However, compatibility with multiple Wi-Fi clients may vary across drivers, so this feature would require thorough testing.

If you're interested in contributing, feel free to submit a pull request!

goatzillax, 2024-11-14:
I think BusyBox should have enough to do what I need.  I think we just need to turn on dumpleases and ntpd server.

Unfortunately I can't seem to get the build system to work right so I'm having trouble testing this, and I don't want to do a PR on something I haven't tested.

https://github.com/goatzillax/thingino-firmware/commit/35025764d4bdc956016852676e64b6e5979a49bd

Is there a way I can get a build with these turned on?

themactep, 2024-11-14:
clone https://github.com/themactep/docker-worker
build inside the container

goatzillax, 2024-11-19:
OK so I'm part of the way there.

Patch to launch NTPD in listener mode for AP:

https://github.com/goatzillax/thingino-firmware/commit/f94cd33f7f93ef97682aaf6b6b3d3316066355cb

Add hostname to dnsd.conf in AP mode and prep for adding clients:

https://github.com/goatzillax/thingino-firmware/commit/a187f152e2878a50d89aeef918698515e96077c5

There are two things I'm not super sure about:

1.  There seems to be some nsswitch or whatever issue where the client cameras aren't resolving the hostnames consistently correctly.
2.  When udhcpd assigns a new lease it can dumpleases into /tmp/dnsd.conf, but what would be the preferred way to restart the dnsd inside a script?