themactep/thingino-firmware #319: Web UI Streamer page corrupts prudynt.cnf with concurrent updated

foxey, 2024-11-13:
The web UI component updates `/etc/prudynt.cfg` for each change in the front-end form. Rapid successive updates cause the file to be truncated and lets the subsequent reload/restart of prudynt fail.

Workaround: reboot the device.

foxey, 2024-11-13:
An example of a truncated file is found in this [diagnostic log](https://termbin.com/bvan).

tERB00, 2024-11-13:
I can confirm this is an issue.

foxey, 2024-11-13:
I briefly read through the code and my conclusion is that the Web UI is sending websocket commands to prudynt and doesn't read/write `/etc/prudynt.cfg'. It is therefor probably an upstream bug in prudynt itself.

themactep, 2024-11-15:
I have changed web ui to not request config updates with every parameter change. Now you have to click the button to save all adjustments into the config file. See if it helps.