themactep/thingino-firmware #320: Onvif motion alert does not work

tERB00, 2024-11-14:
Motion detection DOES work. I am receiving telegram snapshots no problem.
Onvif notifications however are not working.

onvif_notify_server is being started at boot with no errors.
/tmp/notify.log
/tmp/onvif.log 
are empty.

onvif.conf has:
```#EVENT
events=3
#Event 0
topic=tns1:VideoSource/MotionAlarm
source_name=VideoSourceConfigurationToken
source_value=VideoSourceToken
input_file=/run/motion/motion_alarm
```

daoudeddy, 2024-11-15:
run ```echo 5 > /tmp/onvif.debug``` and ```tail -F /tmp/onvif.1.log``` to check the logs

tERB00, 2024-11-15:
> run `echo 5 > /tmp/onvif.debug` and `tail -F /tmp/onvif.1.log` to check the logs

There is nothing motion-related in the onvif logs - the program exits like normal.

```
2024-11-15 12:01:46 DEBUG conf.c:122: get_presets: /sbin/ptz_presets -g
2024-11-15 12:01:46 DEBUG conf.c:122: jump_to_abs: motors -d h -x %f -y %f
2024-11-15 12:01:46 DEBUG conf.c:122: jump_to_rel: motors -d g -x %f -y %f
2024-11-15 12:01:46 DEBUG conf.c:122: events: 1
2024-11-15 12:01:46 DEBUG conf.c:122: topic: tns1:VideoSource/MotionAlarm
2024-11-15 12:01:46 DEBUG conf.c:122: source_name: VideoSourceConfigurationToken
2024-11-15 12:01:46 DEBUG conf.c:122: source_value: VideoSourceToken
2024-11-15 12:01:46 DEBUG conf.c:122: input_file: /run/motion/motion_alarm
2024-11-15 12:01:46 INFO  onvif_simple_server.c:294: Completed.
2024-11-15 12:01:46 DEBUG onvif_simple_server.c:308: Malloc finished,input pointer points to 0x772a0030
2024-11-15 12:01:46 DEBUG onvif_simple_server.c:325: Realloc finished,input pointer is now pointing to 0x77419570
2024-11-15 12:01:46 DEBUG onvif_simple_server.c:333: Input:
<?xml version="1.0" encoding="utf-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:wsse="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd" xmlns:wsu="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:wsa5="http://www.w3.org/2005/08/addressing" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:dn="http://www.onvif.org/ver10/network/wsdl" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:tmd="http://www.onvif.org/ver10/deviceIO/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl" xmlns:ter="http://www.onvif.org/ver10/error">
<SOAP-ENV:Header>
              </SOAP-ENV:Header>
      <SOAP-ENV:Body>
        <tds:GetSystemDateAndTime/>
      </SOAP-ENV:Body>
    </SOAP-ENV:Envelope>

2024-11-15 12:01:46 DEBUG onvif_simple_server.c:334: Url: device_service
2024-11-15 12:01:46 DEBUG onvif_simple_server.c:348: Method: GetSystemDateAndTime
2024-11-15 12:01:46 INFO  onvif_simple_server.c:652: Program terminated.
```

How do I enable notify logs?

tERB00, 2025-02-24:
Fixed in https://github.com/themactep/thingino-firmware/commit/530ea69c423cf2ca8a60f7f3be57cb9d639303ff

closed 2025-02-24 by tERB00