themactep/thingino-firmware #322: Special chars in ONVIF password break things in UI (maybe elsewhere?)

numericOverflow, 2024-11-15:
If you enter an ugly password with lots of special chars, the webUI gets very angry.  Main page loads OK, but if you try to browse the "Settings > RTSP/ONVIF Access" page, it times out and will never display.

Using SSH connection to manually revert the password in the `/etc/shadow` `/etc/onvif.conf` and `/etc/prudynt.cfg` files to something with only alphanumerics fixes the issue and things go back to normal.

Example:  Let's be nasty and change the password to `(*&!*)%&^~^@##%`
- Navigate to "Settings > RTSP/ONVIF Access" page
- Change the RTSP/ONVIF  password to `(*&!*)%&^~^@##%`
- Click "Save Changes"
Result:
- Page times out when saving, but seems to update underlying files as expected.
- Main page of UI loads OK, correctly shows new password at bottom below camera image preview
  `mpv --profile=fast rtsp://thingino:(*&!*)%&^~^@##%@webcam13.local:554/ch0`
- Navigating to "Settings > RTSP/ONVIF Access" page will never return

I didn't test to see where exactly it causes issues (by manually changing one of the three shown locations that get updated on password change.  Obviously my example is a particularly antagonistic password meant to test character safety, but you get the idea.

themactep, 2024-11-16:
that's easy: don't use ugly passwords :)

themactep, 2024-11-18:
please test with tomorrow build. should be fixed by then.

themactep, 2024-11-23:
no response means all good :) reopen if still broken.

closed 2024-11-23 by themactep