themactep/thingino-firmware #323: Wyze v3 on UCTRONICS adapter connection issues

tismofied, 2024-11-16:
I have a couple of wyze v3 with thingino. I am using the UCTRONICs poe adapter for power and data. 
They are all connected to a poe switch with a budget of 64w. There is no usb OTG cables in between. If they are all started at the same time, they work for roughly 4 to 5 hours then they start losing connection.
I can’t figure out why they would work for hours then start dropping off network.
The poe adapter still shows active on the router
How do I troubleshoot this issue please?
I would like to know what log to read right before the camera lost connection if possible.

themactep, 2024-11-16:
there are three major loggers in thingino accessible via `dmesg`, `logread`, `logcat`

gtxaspec, 2024-11-17:
Make sure your POE switch actually puts out 64w in the budget... I've had some generic switches that can not sustain the claimed power output for long.

tismofied, 2024-11-17:
Thank you both. The switch I have is [tp-link LS108GP](https://a.co/d/cAwg4RS)
It's a decent unmanaged poe switch. 
I am leaning towards giving up on the poe option altogether and just using wifi. Picture Quality gonna be reduced but it's better than nothing. 
With the adapters, there is too many points of failure and since I am planing on setting these v3s through the attic, I rather have something that just works.
Thanks guys for your suggestions.

gtxaspec, 2024-11-27:
moving this to discussions for visibility