themactep/thingino-firmware #324: motion guard might could use a cork/inhibit during ptz operations

goatzillax, 2024-11-19:
Maybe PTZ can drop a cork or inhibit file in /tmp that the motion script will check for before firing, and then delete the cork when PTZ is finished?

themactep, 2024-11-19:
good idea. @gtxaspec, if you add a pid file (say /run/motors.pid) to `motors` app then i'll make `motion` script to check for that file and bail out if it is set.

themactep, 2024-11-20:
won't that run another copy of motors with all the overhead?

gtxaspec, 2024-11-20:
it will, but it works effectively while keeping things lightweight. the app is tiny and exits once it completes, which eliminates us from adding more bloat in the daemon.  overhead is minimal and won't be noticed.

goatzillax, 2024-11-20:
The app does look tiny; I wonder if there is potential for a race between the intention to move, motor, and the motion script finally getting back the status that the motors are now indeed busy.

Seems like the cork mechanism should be dropped before even calling motor so that the intention can be communicated to the motion script, in order, and as early as possible.

goatzillax, 2024-11-21:
Hrm, I was playing around with wrapping motors with a script that drops a cork for motion look for.

Should motors block until the movement is completed, or would that not be in keeping with design?

themactep, 2025-12-29:
implemented. please test tomorrow build and close the ticket if satisfied with the results.

closed 2026-01-02 by goatzillax