themactep/thingino-firmware #349: Can the pins be figured out from original firmware?

felipecrs, 2024-12-04:
I took a firmware dump of my camera:

[fulldump.bin.txt](https://github.com/user-attachments/files/18015233/fulldump.bin.txt)

(remove .txt suffix after downloading)

I wonder if I can somehow figure out the pins through it.

---

The camera is question is a QJ05T with Ingenic T20L which you helped me to install OpenIPC in 2023.

I have left the camera in the shelf since then, as I was not able to identify the pins lol.

Another question I have: is there a way to go from OpenIPC to thingino?

themactep, 2024-12-05:
@felipecrs ping me on telegram or in discord :) issues are for the code issues.

closed 2024-12-05 by themactep

felipecrs, 2024-12-05:
I'm sorry, you're totally right.

At first I was thinking of naming the issue as "Support for Howell QJ05" to track whatever is needed to get full support for it, including documentation changes.

Anyway, let me ping you there. Thank you very much.