themactep/thingino-firmware #350: RTSP/ONVIF password resets after partial upgrade

RauschUndNot, 2024-12-05:
It seems that the RTSP/ONVIF password resets to the default after partial upgrades: `sysupgrade -p`. 

HARDWARE:
Wyze V3
t31x, gc2053, 16 MB

oopiicaa, 2024-12-05:
Done partial upgrade after a few days - it broke my V3 completely. WGUI is not reachable, cannot SSH, no RTSP. Camera is connected to WiFi though (I can ping it). Will have to investigate when I come home.

themactep, 2024-12-05:
rtsp/onvif password does reset after upgrade bacause overlay gets wiped out.

RauschUndNot, 2024-12-15:
Is it possible to store it where the wifi network name and password are stored?

gtxaspec, 2024-12-15:
preserving overlay between updates is coming soon, you'll know when its ready! =-D

davlaw, 2024-12-16:
Noticed font and font size too, Will be welcome to see!

gtxaspec, 2025-03-03:
overlay is now preserved between updates. please reopen if you still have issues.

closed 2025-03-03 by gtxaspec