themactep/thingino-firmware #353: Wyze Cam v3 (T31AL) Audio Issues

shamlord, 2024-12-06:
5 DEC 2024 firmware. Stream testing done with VLC on Windows. Camera connected to WiFi on same network.

Issues:
Audio cuts out depending on codec. 
Based on the behavior, I suspect the CPU load is too high or audio related processes priority is too low.
- AAC: Nearly constant silence - audio will cut in for a few seconds rarely then cut back out. 
- G726: Rarely cuts out
- PCM: Cuts out slightly less frequently than G726

Audio configuration is inconsistent from the web GUI:
- Setting the codec to AAC with a 16khz sample rate and then accessing the stream with VLC shows the camera sending a Stereo AAC stream at 32khz.
- Setting AAC with an 8000 hz sample rate results in a stereo 16khz stream.
- Setting G726 with a sample rate of 16000 hz results in VLC receiving an 8000 hz G726 stream
   (I think ONVIF may only support a sample rate of 8000 hz for G726)
 - PCM does actually stream at the sample rate it's set to.

CazYokoyama, 2024-12-08:
Are you able to change the title to "Audio cuts out on Wyze Cam v3 (T31AL)"? The people other than you read the title and find there is a problem on audio. They have to read your description to find what problem. My suggested title eliminate reading the description to find what problem.

gtxaspec, 2025-01-04:
please try the latest build as of 01/04 and let us know if it resolves any audio issues.

shamlord, 2025-01-06:
Testing with firmware:
IMAGE_ID=wyze_c3_t31al_atbm
BUILD_ID="master+a29f523, 2025-01-05 23:52:12 -0500"

In the WebUI, clicked Settings > Streamer > Audio
Codec: AAC. Double clicked Sampling for 16000 hz default, double clicked Bitrate for blank default.
Clicked Save config, Restart streamer
Connect to Main RTSP stream with VLC.
Audio codec is still **OPUS, Stereo, 48000 khz 32-bit**. Audio quality sounds normal.
Checked prudynt.cfg:
>input_enabled = true;
  input_high_pass_filter = true;
  input_agc_enabled = true;
  input_format = "AAC";
  input_bitrate = 40;
  input_sample_rate = 16000;

Reloaded WebUI, Streamer Audio settings show AAC at 16kHz.
WebUI Tools > Reboot.
Reconnect to main stream with VLC.
Codec now correctly shows AAC, 16000 khz.  There is no audio.
VLC Statistics tab shows only video blocks receeived, and the count increases as the stream continues. Zero audio blocks Decoded, Played, or Lost.
Stopped VLC stream playback.
In the WebUI, clicked Services > Streamer > Audio.
Changed the codec to G726. Double clicked Sampling for 16000khz default, double clicked Bitrate for blank default.
Clicked Save config. Waited 5 seconds. Clicked Restart Streamer.
Reconnected to main RTSP stream with VLC.
Audio codec is unchainged. AAC, 16kHz. Still no audio blocks received.
Stopped playback in VLC.
In the WebGUI, clicked Tools > Reboot.
In the WebGUI, checked Streamer Audio settings. 
Codec reflects G726, 16kHz Sampling, Bitrate 40 kbps.
Reconnected to RTSP main stream with VLC.
VLC reports stream audio codec as G.726 ADPCM, Mono, 8000 hz sample rate, 16-bit.
VLC statistics show audio blocks being received and decoded.
Sound quality is much worse than OPUS.  Noise is heard but nothing intelligible.
Allowed VLC stream playback to continue but muted audio.

For my personal reference, listened to G.726 audio samples at: https://vocal.com/audio-examples/itu-speech-coder-audio-examples/
Decided to unmute VLC audio stream again, found that the audio was much improved with no configuration changes by me. Sounds picked up by the camera mic are clear and intelligible. Perhaps the audio quality is poor for a short time after camera boot due to CPU load or RAM usage?
Rebooted camera again via WebUI.
Audio is just noise once again. Waited.
With the system uptime at around 2m 50s, the audio changed drastically, starting out very loud but then decreasing in volume to a reasonable level. Sound was once again intelligible and reasonable quality for G726.
Checked Audio section in prudynt.cfg again.
>  input_enabled = true;
  input_high_pass_filter = true;
  input_agc_enabled = false;
  input_format = "G726";
  input_bitrate = 40;
  input_sample_rate = 16000;
  input_vol = 120;
  input_gain = 30;
  input_alc_gain = 7;
  input_agc_target_level_dbfs = 30;
  input_agc_compression_gain_db = 70;
  input_noise_suppression = 3;

CFG accurately reflects WebGUI.

Results:
AAC appears to still be broken.
G.726 is working, but does take up to 3 minutes after boot for audio stream to be intelligible.
Changes to the WebGUI / prudynt.cfg do not take effect when restarting Streamer. Changes only work after a reboot.

shamlord, 2025-01-06:
After reading back through my testing notes, I decided to retry AAC after giving the camera several minutes to finish starting.

In the WebGUI, went to Streamer Audio settings.
Set codec to AAC and double clicked the Sampling and Bitrate fields to set them to default (16000 Hz, 40 kbps).
Saved streamer config and then rebooted the camera via WebGUI.
After the reboot, reconnected to the main RTSP stream at about 1 minute of uptime.
VLC now shows audio blocks being received, but no sound. Waited. 
At approximately 2 minutes of uptime, I now hear audio, at a much higher quality than G.726.
The AAC stream reported by VLC now shows Stereo, 32000 Hz, 32-bit.
Checked the web GUI. It shows audio codec set to AAC, 16000 Hz, 40 kbps.
Checked prudynt.cfg and found that it matches the WebGUI:
>  input_enabled = true;
  input_high_pass_filter = true;
  input_agc_enabled = false;
  input_format = "AAC";
  input_bitrate = 40;
  input_sample_rate = 16000;
  input_vol = 120;
  input_gain = 30;
  input_alc_gain = 7;
  input_agc_target_level_dbfs = 30;
  input_agc_compression_gain_db = 70;
  input_noise_suppression = 3;

gtxaspec, 2025-01-06:
thx for the info.  i'll do some testing locally as well.

schmidtch89, 2025-01-07:
I notice also an issue with the audio with my two cinnado_d1_t23n.
With one of the last releases the audio stream via RTSP is not functional anymore.
Looks like this issue is not only related to the camera model mentioned in the header.

gtxaspec, 2025-01-11:
some fixes to the streamer coming with audio fixes... stay tuned.

gtxaspec, 2025-01-13:
please try the latest build, there has been updates to the streamer as of todays build.

gtxaspec, 2025-01-18:
@shamlord is this still a problem? let us know!

schmidtch89, 2025-01-18:
Hey @gtxaspec , for my Cinnado D1 audio is working again. But it is not synchronized with the video. There is a delay of approx. one or two seconds.

shamlord, 2025-01-19:
IMAGE_ID=wyze_c3_t31al_atbm
BUILD_ID="master+f4d8d71, 2025-01-19 02:37:03 -0800"
BOOTLOADER=isvp_t31al_sfcnor_ddr128M

Upgraded to the firmware above via partial upgrade using Flash Operations in Web GUI.
Firmware defaults to OPUS, 16000hz, 40kbps in web GUI.
prudynt.cfg audio setting matches web GUI.
Connected to Main stream with VLC.
Audio codec reported by VLC is OPUS, Stereo, 48000hz, 32 bit sample rate

Using Web GUI, changed audio codec to G726, 8000 hz, 40 kbps.
Confirmed change shows in prudynt.cfg.
Clicked Restart Streamer. No change in VLC.
Stopped VLC playback.
Clicked Restart Streamer.
Started VLC stream playback. Audio codec is still OPUS.
Restarted Camera via WebGUI.
Resumed playback of camera stream in VLC at around 1m 30s of uptime.
Little or no audio. Waited.
At 2m 54s of uptime, the audio changed and I could now hear sound.
Automatic gain control initially adjusted the gain up, then back down to a reasonable level.
VLC shows codec is G726, Mono, 8000 hz. 
I can hear clear audio (for G726).

In WebGUI, changed audio codec to AAC, 16000hz, 40 kbps.
Clicked Save Config.
Confirmed change is in prudynt.cfg.
Clicked Restart Streamer.
Disconnected and reconnected to stream with VLC.
VLC still reports G726, 8000hz mono.
Rebooted camera in WebGUI.

Reconnected to video stream at 45s of uptime (as soon as the WebGUI came back up).
VLC reports AAC @ 16000 hz, but only 1 audio block was received.
Disconnected and reconnected to stream with VLC.
VLC reports audio codec is still AAC at 16000 hz, but no blocks of audio are being received.
Stopped playback in VLC and waited.
Reconnected to stream at 4m 15s. Still no audio.
VLC started receiving audio blocks at approximately 4m 45s of uptime.
Audio is very clear and intelligible. 
Audio codec reported by VLC has change to AAC, at 32000 hz, 32 bit sample rate.
Confirmed Web GUI and prudynt.cfg still show:

>   input_format = "AAC";
>   input_bitrate = 40;
>   input_sample_rate = 16000;

Streaming for 17m continuously, AAC has not yet cut out.
Rebooted camera in web gui.
Did not stop VLC, it displayed an error at disconnect.
VLC automatically reconnected to the stream at approximately 30 seconds of uptime (before WebGUI came back up)
VLC shows it was receiving AAC audio blocks immediately.
Confirmed clear audio at about 1m of uptime.
The codec reported by VLC is Stereo, 32000hz, 32 bit sample rate.
Confirmed Web GUI and prudynt.cfg are still configured to 16000hz AAC.

Conclusion:

- AAC now works, although the encoder still doesn't respect the settings in prudynt.cfg. 
- AAC and OPUS both stream in stereo at frequency rates other than what is set in prudynt.cfg. The overall stream bitrate would be lower if the audio were mono and the frequency settings were honored. Unless the camera has multiple microphones I'm not aware of. I suspect it may also cut down on the CPU load from the encoder process.
- I usually enable the camera reboot entry in crontab, but I'll leave it commented out and check on the camera daily to verify stability.
- Unrelated, but restarting streamer has no apparent use if streamer only applies audio settings at reboot. However, the OSD options seem to update right away without a streamer restart when those settings are adjusted.

benjani85, 2025-04-29:
I also have a Wyze Cam V3 where the audio doesn't work on the ONVIF stream although I can hear audio from RTSP.

I have the main and substream set to AAC codec. What am I doing wrong?

Thingino build: master+55a3254, 2025-04-27 09:46:35 +0000

shamlord, 2025-07-20:
I haven't messed with my test camera in a while, since I tried a WebUI full upgrade a few months back and then was too busy (or lazy) to go grab it off the wall and reconnect it to my network. I checked on the status of this project today and did a full flash of the firmware from 16 July:
> SOC=t31
> SOC_ARCH=xburst1
> IMAGE_ID=wyze_cam3_t31al_gc2053_atbm6031
> BUILD_ID="master+596d148, 2025-07-16 09:12:26 +0000"
> BUILD_TIME="2025-07-16 09:12:26 +0000"
> COMMIT_ID="master+596d148, 2025-07-15 20:00:42 +0000"
> BOOTLOADER=isvp_t31al_sfcnor_ddr128M

So far, it's fantastic. I had to switch the audio codec from OPUS to AAC for my NVR of course, but I did not have to do anything with prudynt.cfg, I did not have to make any adjustments to the sample rate either. Also, the Streamer restart button in the WebUI worked, and the audio configuration changes were applied without having to reboot the camera. 

The only "issue" I've noticed is that the WebUI shows a configured sample rate of 16,000hz, yet the stream data reported by VLC shows 32,000hz (Stereo). Not a big deal as far as I'm concerned. 

If you want to look further into this Mono mic vs Stereo stream issue, here is a section from prudynt.cfg from my camera:
> audio : 
> {
>   input_enabled = true;
>   output_enabled = false;
>   force_stereo = false;
>   input_high_pass_filter = false;
>   input_agc_enabled = false;
>   input_format = "AAC";
>   input_bitrate = 40;
>   input_sample_rate = 16000;
>   output_sample_rate = 16000;
>   input_vol = 80;
>   input_gain = 25;
>   input_alc_gain = 0;
>   input_agc_target_level_dbfs = 10;
>   input_agc_compression_gain_db = 0;
>   input_noise_suppression = 0;

And here is what the VLC Codec tab reports:
> Codec: MPEG AAC Audio (mp4a)
> Type: Audio
> Channels: Stereo
> Sample rate: 32000 Hz
> Bits per sample: 32

Even if this stuff got fixed in response to another bug, or maybe just an improvement or fix you folks found on your own, the current version is awesome. If I find it's stable enough on one of my cameras I may just flash all the rest of them. I'll follow up on this issue in a bit once my camera has had some uptime.