themactep/thingino-firmware #355: Building on WSL fails because of PATH

felipecrs, 2024-12-07:
`PATH` in WSL is a concatenation of Linux' PATH with Windows' PATH. Example:

```
/home/felipecrs/.local/bin:/home/felipecrs/bin:/home/felipecrs/go/bin:/home/felipecrs/.deno/bin:/home/felipecrs/.volta/bin:/home/linuxbrew/.linuxbrew/bin:/home/linuxbrew/.linuxbrew/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/lib/wsl/lib:/mnt/c/Program Files/Volta/:/mnt/c/Program Files/Eclipse Adoptium/jre-21.0.5.11-hotspot/bin:/mnt/c/Program Files/OpenSSH/:/mnt/c/Windows/system32:/mnt/c/Windows:/mnt/c/Windows/System32/Wbem:/mnt/c/Windows/System32/WindowsPowerShell/v1.0/:/mnt/c/Windows/System32/OpenSSH/:/mnt/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/mnt/c/Windows/system32/config/systemprofile/AppData/Local/Microsoft/WindowsApps:/mnt/c/Program Files/dotnet/:/mnt/c/WINDOWS/system32:/mnt/c/WINDOWS:/mnt/c/WINDOWS/System32/Wbem:/mnt/c/WINDOWS/System32/WindowsPowerShell/v1.0/:/mnt/c/WINDOWS/System32/OpenSSH/:/mnt/c/Program Files (x86)/Windows Kits/10/Windows Performance Toolkit/:/mnt/c/Program Files/WireGuard/:/mnt/c/Program Files/NVIDIA Corporation/NVIDIA App/NvDLISR:/mnt/c/Program Files/gsudo/Current:/mnt/c/Program Files/Git/cmd:/mnt/c/Program Files/Go/bin:/mnt/c/Program Files/PuTTY/:/mnt/c/Program Files/GitHub CLI/:/mnt/c/Users/felip/AppData/Local/Programs/Python/Python312/Scripts/:/mnt/c/Users/felip/AppData/Local/Programs/Python/Python312/:/mnt/c/Users/felip/AppData/Local/Programs/Python/Launcher/:/mnt/c/Users/felip/.cargo/bin:/mnt/c/Users/felip/AppData/Local/Volta/bin:/mnt/c/Users/felip/AppData/Local/Microsoft/WindowsApps:/mnt/c/Users/felip/AppData/Local/Programs/oh-my-posh/bin:/mnt/c/Users/felip/AppData/Local/Microsoft/WinGet/Links:/mnt/c/Users/felip/AppData/Local/Programs/Microsoft VS Code/bin:/mnt/c/Users/felip/AppData/Local/Microsoft/WindowsApps:/mnt/c/Users/felip/AppData/Local/Programs/Komac/bin:/mnt/c/Mtl-Multisync/Rewriters/_msrgallwinner.zip/:/mnt/c/Users/felip/AppData/Local/Microsoft/WinGet/Packages/PolarGoose.BluetoothDevicePairing_Microsoft.Winget.Source_8wekyb3d8bbwe/:/mnt/c/Users/felip/AppData/Local/Microsoft/WinGet/Packages/Google.PlatformTools_Microsoft.Winget.Source_8wekyb3d8bbwe/platform-tools:/mnt/c/Program Files/Multipass/bin:/mnt/c/Users/felip/go/bin:/mnt/c/Program Files (x86)/Nmap
```

And `make` fails with:

```
Your PATH contains spaces, TABs, and/or newline (\n) characters.
This doesn't work. Fix you PATH.
```

I wonder if this requirement can be alleviated.

The workaround is to add:

```
[interop]
appendWindowsPath = false
```

To `/etc/wsl.conf`.

themactep, 2024-12-09:
do not build in wsl.
https://github.com/themactep/thingino-firmware/wiki/Development#containers

gtxaspec, 2024-12-09:
the space in the path issue is with the buildroot system itself, we added a check in the top level Makefile to prevent people from running into errors during the build... do not build in WSL =D
If you're on windows you could try the docker container that @themactep linked above... ymmv, i dont use windows lol

felipecrs, 2024-12-09:
Well, the workaround I mentioned worked just fine. I was able to build the firmware multiple times on WSL.

But I'll check this container out.

And if it's a limitation of buildroot, then there's nothing we can do indeed. At least the workaround is now documented.

Thank you!

closed 2024-12-09 by felipecrs

themactep, 2024-12-09:
we use a patched buildroot so if you figure out a fix we can add it to our buildroot repo.