themactep/thingino-firmware #361: Bin download for partial upgrade failed

weromemuero, 2024-12-11:
When I try to do a partial upgrade I get the following error:

Partial upgrade from GitHub
Upgrading from GitHub
Download firmware and checksum
- https://github.com//themactep/thingino-firmware/releases/download/firmware_update/thingino-wyze_c3_t31x_rtl-update.bin
- https://github.com//themactep/thingino-firmware/releases/download/verify/thingino-wyze_c3_t31x_rtl-update.bin.sha256sum
curl -LO# https://github.com//themactep/thingino-firmware/releases/download/firmware_update/thingino-wyze_c3_t31x_rtl-update.bin
#=#=- #    #
curl: (22) The requested URL returned error: 404

#=#=- #    #
curl: (22) The requested URL returned error: 404

Download failed!

Also when I try to get the firmware from my web browser I get the 404 page not found error.

gtxaspec, 2024-12-11:
you must be running an older build.  the system upgrade mechanism has changed.  you can do this:

Run the following command on the camera:
```curl -sL -o /sbin/sysupgrade https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/overlay/lower/usr/sbin/sysupgrade && chmod +x /sbin/sysupgrade```

then run sysupgrade -p as usual.

themactep, 2024-12-11:
the solution is plastered all over the support channels and in the wiki.

weromemuero, 2024-12-11:
Thanks for the help!