themactep/thingino-firmware #362: Feature: allow to set OSD items position from WebUI

felipecrs, 2024-12-11:
It would be great if I could select the position of the time item of OSD directly from WebUI.

PS: this can be configured already in the prudynt config file: https://github.com/gtxaspec/prudynt-t/wiki/Configuration#osd_pos_x

One suggestion is to add some presets for users to choose from, like one for each corner, maybe one for centering, etc.

felipecrs, 2024-12-11:
BTW, @gtxaspec, `pos_time_x` and `pos_time_y` documentation is missing from the wiki page (it's not open to edit).

gtxaspec, 2024-12-11:
Wiki is open now, thanks for the note on that, didn't realize!

felipecrs, 2024-12-11:
Made this change: https://github.com/gtxaspec/prudynt-t/wiki/Configuration/_compare/4d3e59daffdf2bda42a1dbcd8fb2104ad2c3502c

I think the entire page is a bit outdated, but let's do one step at a time. :D

gtxaspec, 2025-04-01:
hopefully coming soon with upgrades in prudynt.

closed 2026-04-15 by themactep