themactep/thingino-firmware #366: Documentation- Compitation information.

Shayol-sh, 2024-12-12:
I would like to notice that there is a big error in documentation.
https://github.com/themactep/thingino-firmware/wiki/Buildroot
Precisely
**local.fragment and local.mk section**
Documentation states the that file
**config/fragments/local.fragment**
Is being used to add additional packages. However declaration in that file doesn't change anything.
Correct path for that file is in main directory. And then additional packages can be added.
[40e02c2](https://github.com/themactep/thingino-firmware/commit/40e02c205dbd306cae4ca375485f96624c5774d9)
**configs/fragments/local.fragment > ‎local.fragment**
Seams that change was not reflected in documentation, so I lost few hours on that.

Second thing about documentation.
There is no information how to use "the Configuration Menu". You can select packages there and kernel modules and options with menuconfig. As a result we can get a .Config file with custom settings. But it is being overwritten every time when compiler starts. Overwritten- Assuming suggested save location (output/device/.Config). Is there any way to force compiler to use that .Config file ? That matter is not explained in documentation at all. On the other hand Buildroot documentation is very complex.

themactep, 2024-12-12:
You've spent so many letters writing the issue instead of just editing the page and correcting the path of `fragment.local`? OK, I fixed the page.

For saving your changes to the camera config use `br-savedefconfig` or a shorthand [`saveconfig`](https://github.com/themactep/thingino-firmware/blob/9b102b2eec8f152e21b6b4d7294bf35196947ad8/Makefile#L197)

Shayol-sh, 2024-12-12:
I didn't know that I can edit documentation page without approval etc.

closed 2024-12-12 by Shayol-sh