themactep/thingino-firmware #373: Curl is missing in the 2024-12-10 build

Bartvz, 2024-12-13:
Since I am a serial updater of all my devices I ran `sysupgrade -p` on my Cinnado D1 t23n  which was on the 2024-12-10 build (BUILD_ID="master+2d66d9d, 2024-12-10 05:02:27 -0500") and got the output which I pasted below.
`root@Cinnado-D1 sbin# sysupgrade -p
Run as /sbin/sysupgrade -p
Updating self
/sbin/sysupgrade: line 69: curl: not found
chmod: /tmp/sysupgrade/sysupgrade: No such file or directory
/sbin/sysupgrade: line 69: curl: not found
Re-running script from /tmp/sysupgrade
/sbin/sysupgrade: exec: line 248: /tmp/sysupgrade/sysupgrade: not found`

Curl seems to be missing. wget is included but won't download from https.

davlaw, 2024-12-13:
Was coming to look up this issue myself for a wyze v2 that I mess up the other day and nothing would work to recover it.  So I cracked it open and did the clone tool on it and did get it recovered. Then discovered this same issue.  Guess I will be following this thread for best recover\y options as Thingino is new to me, have only discovered it a few days ago ;) 

NAME=Thingino
ID=thingino
VERSION="1 (Ciao)"
VERSION_ID=1
VERSION_CODENAME=ciao
PRETTY_NAME="Thingino 1 (Ciao)"
ID_LIKE=buildroot
CPE_NAME="cpe:/o:thinginoproject:thingino:1"
LOGO=thingino-logo-icon
ANSI_COLOR="1;34"
HOME_URL="https://thingino.com/"
ARCHITECTURE=mips
IMAGE_ID=wyze_c2_jxf23
BUILD_ID="master+2d66d9d, 2024-12-10 05:02:27 -0500"
HOSTNAME=ing-wyze-c2
TIME_STAMP=1733706349
BUILDROOT_NAME=Buildroot
BUILDROOT_VERSION=-g791cb854
BUILDROOT_ID=buildroot
BUILDROOT_VERSION_ID=2025.02-git
BUILDROOT_PRETTY_NAME="Buildroot 2025.02-git"
 
Run as /sbin/sysupgrade -p
Creating /tmp/sysupgrade directory
Updating self
/sbin/sysupgrade: line 69: curl: not found
chmod: /tmp/sysupgrade/sysupgrade: No such file or directory
/sbin/sysupgrade: line 69: curl: not found
Re-running script from /tmp/sysupgrade
/sbin/sysupgrade: exec: line 248: /tmp/sysupgrade/sysupgrade: not found

davlaw, 2024-12-14:
Thanks, that was easy enough.  Thinino is a recent adoption in my arsenal, thanks for the help with a "noob" question

themactep, 2024-12-15:
[curl.gz](https://github.com/user-attachments/files/18139930/curl.gz)

1. download
2. gunzip curl.gz
3. scp -O curl root@cameraip:/bin/
4. ssh root@cameraip chmod +x /bin/curl

Profit!

gtxaspec, 2024-12-15:
moving to discussions for visibility as this is already fixed

closed 2024-12-15 by gtxaspec