themactep/thingino-firmware #376: RTSP urls on rtsp settings page showing incorrectly with double quotes surrounding channel name

savingprivatebryan, 2024-12-14:
Small issue, the Mainstream and Substream urls on the RTSP/ONVIF Access settings page in the web ui has double quotes in the channel name but the actual endpoints don't have the double quotes. 
e.g. it shows rtsp://thingino:thingino@192.168.1.45:554/"ch0" which doesn't work unless you remove the double quotes.

[image]

This was using wyze_c3_t31al_atbm

gtxaspec, 2024-12-20:
is this still an issue in the latest builds?

Quadpilotman, 2024-12-22:
This also occurs on the wuuk Y0510 just on the substream URL

themactep, 2024-12-30:
fixed in https://github.com/themactep/thingino-firmware/commit/baf2b561a1245d2fb6fa0d98022c79975738e59b

closed 2024-12-30 by themactep