themactep/thingino-firmware #378: Tapo C100 v5 supported?

ikout, 2024-12-14:
Hello, I got a new Tapo C100 but the only available was a v5 EU. Is this device going to work?
When I look inside I see that the main CPU is a T23 and the Wifi chip seem to be different..

Thingino site says for C100: "T31L, SC2336, RTL8188FTV, 8MB"
The device I have is based on the T23 CPU and for WiFi it seems to have a WQ9001 chip...

Any ideas if this device will be possible to be flashed with Thingino? And which settings???

gtxaspec, 2024-12-14:
Cpu is supported, wifi not yet... We are missing the driver sources.

ikout, 2024-12-14:
So can't flash it then.... :( 
That's unfortunate, and I really wanted to test Thingino...

themactep, 2024-12-15:
Buy a supported camera from the list at http://thingino.com/
Say, Cinnado D1 is available in EU.

themactep, 2024-12-15:
Guys, you want new hardware supported - send a specimen to developers. We cannot work on models that we do not have on hands.

gtxaspec, 2024-12-24:
@ikout initial support has been added for the WQ9001 wifi chip.  Would you please confirm that your model has this chip?

ikout, 2024-12-24:
Hi @gtxaspec, 

I believe that that's the one we are talking about, so yes.

[image]

gtxaspec, 2024-12-24:
@ikout do you have a dump of the stock firmware?

ikout, 2024-12-24:
No, I've not tried anything yet since there was no support on this device in general. 

I think I've seen 4 pads that maybe for serial or USB. I don't mind to brick temporarily, I got this one to use Thingino anyway (if possible)

gtxaspec, 2025-01-03:
see:
https://github.com/themactep/thingino-firmware/commit/5d0e8d748891ec39d673722e96981a0c18333748

gtxaspec, 2025-01-03:
added

closed 2025-01-03 by gtxaspec